timfpark/react-native-location #8: NativeModule syntax no longer valid?

chris-erickson, 2015-11-12:
I'm unable to build this in it's current state, with RN `0.14.0`.  I get:

```
bundle: Created ReactPackager
uncaught error Error: UnableToResolveError: Unable to resolve module NativeModules from {MY_PROJECT_PATH}/node_modules/react-native-location/RNLocation.ios.js: Invalid directory /Users/node_modules/NativeModules
```

When changing [this line](https://github.com/timfpark/react-native-location/blob/master/RNLocation.ios.js#L7): 

```javascript
var NativeRNLocation = require('NativeModules').RNLocation;
```
to this:
```javascript
var NativeRNLocation = require('react-native').NativeModules.RNLocation;
```
Things start working again.  It seems vaguely related [to this](https://github.com/corymsmith/react-native-icons/issues/59).

nicinabox, 2015-11-13:
#9 should address this.

chris-erickson, 2015-12-03:
Works great now with 0.15.0.

closed 2015-12-03 by chris-erickson