timfpark/react-native-location #13: Trying to test in background

ButuzGOL, 2016-01-18:
On simulator background works find but on device I puts console.logs but they not triggered
May you can help me ?

ButuzGOL, 2016-01-18:
My code
console.log not triggered 
```
Location.requestAlwaysAuthorization();
Location.startUpdatingLocation();
Location.setDistanceFilter(5.0);

var subscription = DeviceEventEmitter.addListener(
    'locationUpdated',
    (location) => {
      console.log('updated');
    }
);
```

timfpark, 2016-01-19:
Hi @ButuzGOL:  Are you using react-native v0.14 or greater?  I believe background JavaScript event support was added as of v0.14.

ButuzGOL, 2016-01-19:
I am using ^0.18.0-rc

gasperz, 2016-02-23:
For me it is also not working on real device. Testing with RN 0.21.0-rc1. But working on simulator.

gasperz, 2016-02-23:
I have just discovered that another function is exposed in this implementation:
`Location.setAllowsBackgroundLocationUpdates(true);`
With adding this code, it is indeed working in background mode. But I think it was added only in iOs9 [(link)](https://developer.apple.com/library/ios/documentation/CoreLocation/Reference/CLLocationManager_Class/#//apple_ref/occ/instp/CLLocationManager/allowsBackgroundLocationUpdates)
So I guess also documentation (about setAllowsBackgroundLocationUpdates) should be update.

ButuzGOL, 2016-02-23:
thanks I will try

closed 2016-03-02 by ButuzGOL