timfpark/react-native-location #14: Implementation in Android

agrass, 2016-02-17:
Hello, 

is there any implementation with Android and Ios support? Are anyone working on this or the implementation still propesed? 

Thanks!

gasperz, 2016-02-23:
For android it might work just Geolocation Api from React-Native. Try with RN 0.20.0 (and more). It works if application is in background.

romanchukenator, 2016-02-25:
Is this confirmed? RN 0.20.0 is the one? It's the version to tie it all together? Anyway, I'll have a look and play around to see if it just works. Pretty sweet if it does.

agrass, 2016-02-25:
I already have in another repository: [](https://github.com/agrass/react-native-gps). Some methods are working for android and IOS. Is not ready for PR to this repo because I adapted some methods to another Android solution that I found.

romanchukenator, 2016-02-25:
Hrm. So am I missing something? While RN-iOS works while app is in the background RN-Android is still stuck relegated to being used only during times the app is active, right? Or is someone saying that there's a solution for background processes in Android?

gasperz, 2016-02-26:
For Android it should work without this module. You can just use [Geolocation Api](http://facebook.github.io/react-native/docs/geolocation.html#content). I tried with RN 0.20.0 (and up) version.

agrass, 2016-02-28:
Great I didn't know about that, Thanks

closed 2016-02-28 by agrass