timfpark/react-native-location #17: Doesn't seem to work with latest react native.

joegaudet, 2016-06-27:
Using the latest react native, the react native location does not appear to get installed on NativeExtensions.

timfpark, 2016-06-30:
Which RN version are you using?  I'm using it with react-native 0.27 and it seems to work.

caledhwa, 2016-07-02:
@joegaudet - I got this to work just fine with v0.28. I've submitted a PR to make installation easier using `react-native link` or `rnpm link` - this properly links your library. However, I found I needed to add 

```xml
	<key>NSLocationWhenInUseUsageDescription</key>
	<string>This is the plist item for NSLocationWhenInUseUsageDescription</string>
	<key>NSLocationAlwaysUsageDescription</key>
	<string>This is the plist item for NSLocationAlwaysUsageDescription</string>
```

In order to get location data back. Other than that, the library seems to work great with v0.28 for me.

caledhwa, 2016-07-04:
@joegaudet - Give the library a try with v0.28 by using the Example app added in PR #20 - once it's merged. Should work great [does for me].

JayChetty, 2017-01-26:
Hi @joegaudet ,  I have same problem using react-native 0.40 + react-native-location 0.27.  Did you find a solution?

romansky, 2017-01-26:
Same here, RN 0.40, cant get it to work..

Filipvdb321, 2017-02-13:
Having the same problem with RN 0.41

romansky, 2017-02-13:
I ended up using native google play code, pasting below for whoever is willing to adopt
```java
public class MyLocation extends ReactContextBaseJavaModule {

    GoogleApiClient client = null;

    @ReactMethod
    public void getLocation(final Callback errorCallback, final Callback successCallback) {

        client = new GoogleApiClient.Builder(this.getCurrentActivity())
                .addConnectionCallbacks(new GoogleApiClient.ConnectionCallbacks() {
                    @Override
                    public void onConnected(@Nullable Bundle bundle) {
                        Location lastLocation = LocationServices.FusedLocationApi.getLastLocation(client);
                        successCallback.invoke(lastLocation.getLatitude(), lastLocation.getLongitude());
                    }
                    @Override
                    public void onConnectionSuspended(int i) {}
                })
                .addOnConnectionFailedListener(new GoogleApiClient.OnConnectionFailedListener() {
                    @Override
                    public void onConnectionFailed(@NonNull ConnectionResult connectionResult) {
                        errorCallback.invoke(connectionResult);
                    }
                })
                .addApi(LocationServices.API)
                .build();

        client.connect();

    }

    public MyLocation(ReactApplicationContext reactContext) {
        super(reactContext);
    }

    @Override
    public String getName() {
        return "MyLocation";
    }
}
```

matt-oakes, 2018-11-14:
Tracking in #35

closed 2018-11-14 by matt-oakes