timfpark/react-native-location #19: Location manager failed

KamranGhoochanian, 2016-07-04:
Location manager failed: Error Domain=kCLErrorDomain Code=0 "(null)"
CFNetwork SSLHandshake failed (-9806)
NSURLSession/NSURLConnection HTTP load failed (kCFStreamErrorDomainSSL, -9806)

i get this error for the first time after calling the Location.startUpdatingLocation() but after a few seconds 'locationUpdated' listener get the right coordinate!
is there any problem?

and is there any sort of callback for requestWhenInUseAuthorization method to know when to call startUpdatingLocation
because i want to get user location after a button pressed not after componentDidMount()

KamranGhoochanian, 2016-07-04:
and my react version is 0.24.1

bnash2501, 2019-09-13:
I forked and wrote the functionality to get the Location Services status for the iOS part. 
However, react-native-permissions added the functionality to get the status for both iOS and Android.