timfpark/react-native-location #24: What is the difference compared to Geolocation ?

note89, 2016-10-26:
Hi ! 
I'm intressted to know the benefits of this compared to https://facebook.github.io/react-native/docs/geolocation.html

We are currently using that in our app, but its very slow on android compared to IOS.

ButuzGOL, 2016-10-29:
+

LeBlaaanc, 2016-11-12:
@note89 your question is kind of off topic because this library doesn't even support Android, however, I would take a look at https://github.com/airbnb/react-native-maps and how they implement tracking, it should provide the same or similar data.

kasemyassine, 2017-01-14:
I want to add to his question is this library more accurate? or it's just the same because I want an accurate gps location functions!

note89, 2017-02-10:
Well if we just talk about IOS then. 
What are the benefits ? :)

skadavan, 2018-01-23:
Same question. How is this project different from : https://facebook.github.io/react-native/docs/geolocation.html ?
Is it better in accuracy or power consumption?

commit 3c9312f4 (2018-11-17): Added rational for the library existing

Closes #24
closed 2018-11-17 by matt-oakes