timfpark/react-native-location #29: Duplicate interface definition for class 'RCTBridge'

Filipvdb321, 2017-02-18:
After doing react-native link I get in xcode the following compilation error:

> Duplicate interface definition for class RCTBridge

If I remove the following line from RNLocation.m it works:
`#import "RCTBridge.h"
`

I am using react native version 0.41.2. Could it be related to this version?

juellez, 2017-02-18:
@Filipvdb321 I had a similar issue, as the version of react-native-location on NPM does not yet have the updated React namespaces. As of RN 40, all imports must be called like `#import <React/RCTBridge.h>` When I pulled the latest version of react-native-location from github, it worked. So I think this is a duplicate of #28 -- and you can see the fix made in https://github.com/timfpark/react-native-location/commit/e67552d7882a0e6664d2cc2912b424181d5b36db

Here's the change you can make to your `package.json`

```
    "react-native-location": "https://github.com/timfpark/react-native-location.git#master",
```

Filipvdb321, 2017-02-19:
Thanks for the info @juellez . It seems that the issue is indeed solved in the latest version! 
I will close this one!

closed 2017-02-19 by Filipvdb321