timfpark/react-native-location #33: undefined is not an object (evaluating '_NativeModules.RNLocation.requestAlwaysAuthorization')

guoliang1206, 2017-04-12:
I do the npm install --save react-native-location ,and react-native link
this is my code , it alert me that undefined is not an object
```
import { RNLocation as Location} from 'NativeModules';
 componentWillMount() {

        Location.requestAlwaysAuthorization();
        Location.startUpdatingLocation();
        Location.setDistanceFilter(5.0);
        DeviceEventEmitter.addListener('locationUpdated', (location) => {
            console.log('^^^^^^^^^^^^^^^^',location);
        })
    }
```

guoliang1206, 2017-04-12:
@timfpark  what's wrong with my code?

ignivaankushrishi, 2017-05-11:
Facing same issue

j-nolan, 2017-06-14:
Could you be using the library with Android?

guoliang1206, 2017-06-15:
@j-nolan no, in fact I give up this lib. but this is still worth. you could learn from this lib and encapsulate your own lib.

closed 2017-08-17 by guoliang1206