timfpark/react-native-location #35: errors with react-native 0.44.0

pjpsoft, 2017-05-14:
setup a clean project

{
  "name": "NatLoc",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest"
  },
  "dependencies": {
    "react": "16.0.0-alpha.6",
    "react-native": "0.44.0",
    "react-native-location": "^0.27.0"
  },
  "devDependencies": {
    "babel-jest": "20.0.1",
    "babel-preset-react-native": "1.9.2",
    "jest": "20.0.1",
    "react-test-renderer": "16.0.0-alpha.6"
  },
  "jest": {
    "preset": "react-native"
  }
}

installed react-native-location
linked
and compile in xcode
gives errors (just showing a few)

../react-native/React/Base/RCTBridge.h:66:12: note: previous definition is here
@interface RCTBridge : NSObject <RCTInvalidating>
           ^
                                                      ^
@property (nonatomic, copy, readonly) NSArray<Class> *moduleClasses;
                                                      ^
: error: property has a previous declaration
@property (nonatomic, strong, readonly) NSURL *bundleURL;
                                               ^
 error: property has a previous declaration
@property (nonatomic, strong) Class executorClass;
                                    ^
In file included from /Users/pierrejean/Dropbox/dev-app/NatLoc/node_modules/react-native-location/RNLocation.m:3:
../react-native/React/Base/RCTBridge.h:166:37: note: property declared here
@property (nonatomic, strong) Class executorClass;
                                    ^
 error: property has a previous declaration
@property (nonatomic, weak, readonly) id<RCTBridgeDelegate> delegate;
                                                            ^
 error: property has a previous declaration
@property (nonatomic, copy, readonly) NSDictionary *launchOptions;
                                                    ^
 error: property has a previous declaration
@property (nonatomic, readonly, getter=isLoading) BOOL loading;
                                                       ^
 error: property has a previous declaration
@property (nonatomic, readonly, getter=isValid) BOOL valid;
                                                     ^
 error: property has a previous declaration
@property (nonatomic, readonly, weak) JSContext *jsContext;
                                                 ^
 error: property has a previous declaration
@property (nonatomic, readonly, strong) RCTPerformanceLogger *performanceLogger;
                                                              ^

^
4 warnings and 10 errors generated.

Build failed    2017-05-14, 12:50 PM

kkroth, 2017-05-24:
I have exactly the same error with these "dependencies": {
    "react": "16.0.0-alpha.6",
    "react-native": "0.44.0",
    "react-native-location": "^0.27.0",
}

rogerluo410, 2017-06-12:
replace #import <React/RCTBridge.h> instead of #import "RCTBridge.h" in project.

I can build successful, but throw out 'undefined is not an Object (evaluating DeviceEventEmitter.addListener)'...

My React-native version is 0.42.3

rexlow, 2017-06-15:
same here, cannot read property of any method :(

kaberibasu, 2017-09-13:
any updates on this issue?