timfpark/react-native-location #40: Get Current Location

arhmnsh, 2019-02-05:
How to get the current location?
`subscribeToSignificantLocationUpdates` will only fire if there is a change in location, what if I just want the current location.
I didn't see any method for that in the documentation.

matt-oakes, 2019-02-05:
When you call `RNLocation.subscribeToLocationUpdates` it will usually call your callback right away with the last known location of the device, however, this is not guaranteed to happen as the device might not have a known location (for example, it has been in Airplane mode since being turned on).

You can subscribe to the events, and then unsubscribe when you receive your first callback. For example:

```javascript
const unsubscribe = RNLocation.subscribeToLocationUpdates(locations => {
  // Do something with the locations

  if (unsubscribe) {
    unsubscribe();
  }
});
```

It would be possible for the library to provide a method which essentially implements this for. I will leave this issue open as a feature request for that and let you know when an update in released.

matt-oakes, 2019-02-06:
This is now included in [v2.1.0](https://github.com/timfpark/react-native-location/releases/tag/2.1.0) which is released on npm.

closed 2019-02-06 by matt-oakes

roysG, 2019-02-18:
Where is the function unsubscribe from the watch?

matt-oakes, 2019-02-18:
@roysG From the code above, calling the `RNLocation.subscribeToLocationUpdates` method returns a function which, when called, will unsubscribe from location updates.

For example:

```javascript
const unsubscribe = RNLocation.subscribeToLocationUpdates(locations => {
  // Do something with the locations
});

// Later on, unsubscribe
unsubscribe();
```

roysG, 2019-02-19:
@matt-oakes , thanks it works.
I have to say, the "subscribeToLocationUpdates" function works amazing in compare to "getCurrentPostion".
I made some tests to check the accurate of the function "getCurrentPostion" and every check the update location is happen immediately just after some 10 meters.

Maybe i miss something? Is there any best practice configuration for this method?

matt-oakes, 2019-02-19:
@roysG If you need to check location every few seconds then you should use `subscribeToLocationUpdates` to subscribe to updates. You can configure that with your desired accuracy and minimum displacement to get the best balance between battery use and accuracy.

The best resources for best practice are the official Apple and Android documentation:

Apple: https://developer.apple.com/documentation/corelocation?language=objc
Android: https://developer.android.com/guide/topics/location/battery

roysG, 2019-02-19:
I read the documentation and i little bit feel confusing.
Lets say you define the next config:

RNLocation.configure({
    distanceFilter: 100, // Meters
    desiredAccuracy: {
      **android: "highAccuracy"**
    }
})

RNLocation.requestPermission({
  android: {
    **detail: "coarse"**
  }
}).then(granted => {
    if (granted) {
      this.locationSubscription = RNLocation.subscribeToLocationUpdates(locations => {
      })
    }
  })

what priority will win?
(As you know coarse is just working with tower and wifi and "highAccuracy" works with gps.)

matt-oakes, 2019-02-19:
The permission levels and desired accuracy are two different things but to get highly accurate locations you will need to requestion "fine" permissions on Android.