timfpark/react-native-location #41: Could not resolve project :react-native-location

johhansantana, 2019-02-09:
Getting this error after manually linking on Android:
```
ERROR: Unable to resolve dependency for ':app@debug/compileClasspath': Could not resolve project :react-native-location.
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugAndroidTest/compileClasspath': Could not resolve project :react-native-location.
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@debugUnitTest/compileClasspath': Could not resolve project :react-native-location.
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@release/compileClasspath': Could not resolve project :react-native-location.
Show Details
Affected Modules: app

ERROR: Unable to resolve dependency for ':app@releaseUnitTest/compileClasspath': Could not resolve project :react-native-location.
Show Details
Affected Modules: app
```

matt-oakes, 2019-02-09:
Hi @johhansantana 

It looks like you have not correctly linked the native part of the library. The instructions for doing that are inside the *iOS* and *Android* sections in the README. Make sure you press the little arrow to open them up and see the full instructions.

I am going to alter the styling of the README to make it more obvious that they are there.

If you have followed the instructions and are still having problems, please open a new issue with the error and the steps to reproduce it.

Thanks!

closed 2019-02-09 by matt-oakes

johhansantana, 2019-02-12:
ah, I did do the linking manually correctly but I had the V1 of this library installed since I was developing iOS first.

Updating the library to the latest version and following the [upgrade guide here](https://github.com/timfpark/react-native-location/releases/tag/2.0.0) fixed the issue