timfpark/react-native-location #42: TypeError: this.nativeInterface.startMonitoringSignificantLocationChanges is not a function

arhmnsh, 2019-02-12:
I'm getting error `TypeError: this.nativeInterface.startMonitoringSignificantLocationChanges is not a function`

**Screenshot:**
[image]

**Version:** react-native-location: ^2.1.1

**Stack trace:**
```
TypeError: TypeError: this.nativeInterface.startMonitoringSignificantLocationChanges is not a function. (In 'this.nativeInterface.startMonitoringSignificantLocationChanges()', 'this.nativeInterface.startMonitoringSignificantLocationChanges' is undefined)

This error is located at:
    in HomeScreen (created by SceneView)
    in SceneView (at StackViewLayout.js:784)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in AnimatedComponent (at StackViewCard.js:69)
    in RCTView (at View.js:44)
    in AnimatedComponent (at screens.native.js:59)
    in Screen (at StackViewCard.js:57)
    in Card (at createPointerEventsContainer.js:27)
    in Container (at StackViewLayout.js:860)
    in RCTView (at View.js:44)
    in ScreenContainer (at StackViewLayout.js:311)
    in RCTView (at View.js:44)
    in AnimatedComponent (at StackViewLayout.js:307)
    in Handler (at StackViewLayout.js:300)
    in StackViewLayout (at withOrientation.js:30)
    in withOrientation (at StackView.js:79)
    in RCTView (at View.js:44)
    in Transitioner (at StackView.js:22)
    in StackView (created by Navigator)
    in Navigator (at createKeyboardAwareNavigator.js:12)
    in KeyboardAwareNavigator (created by SceneView)
    in SceneView (at createTabNavigator.js:39)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in ResourceSavingScene (at createBottomTabNavigator.js:113)
    in RCTView (at View.js:44)
    in ScreenContainer (at createBottomTabNavigator.js:103)
    in RCTView (at View.js:44)
    in TabNavigationView (at createTabNavigator.js:197)
    in NavigationView (created by Navigator)
    in Navigator (created by SceneView)
    in SceneView (created by SwitchView)
    in SwitchView (created by Navigator)
    in Navigator (at createAppContainer.js:388)
    in NavigationContainer (at renderApplication.js:34)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in AppContainer (at renderApplication.js:33)

This error is located at:
    in NavigationContainer (at renderApplication.js:34)
    in RCTView (at View.js:44)
    in RCTView (at View.js:44)
    in AppContainer (at renderApplication.js:33)
subscribeToSignificantLocationUpdates
    subscriptions.js:44:8
componentDidMount
    index.js:45:6
wrapper
    native.js:486:32
fn
    native.js:507:23
commitLifeCycles
    ReactNativeRenderer-dev.js:12144:10
commitAllLifeCycles
    ReactNativeRenderer-dev.js:13496:8
invokeGuardedCallbackImpl
    ReactNativeRenderer-dev.js:92:15
invokeGuardedCallback
    ReactNativeRenderer-dev.js:306:36
commitRoot
    ReactNativeRenderer-dev.js:13702:8
completeRoot
    ReactNativeRenderer-dev.js:15269:13
performWorkOnRoot
    ReactNativeRenderer-dev.js:15197:21
performWork
    ReactNativeRenderer-dev.js:15090:24
performSyncWork
    ReactNativeRenderer-dev.js:15047:14
requestWork
    ReactNativeRenderer-dev.js:14925:19
scheduleWork
    ReactNativeRenderer-dev.js:14711:16
scheduleRootUpdate
    ReactNativeRenderer-dev.js:15429:15
render
    ReactNativeRenderer-dev.js:16142:20
renderApplication
    renderApplication.js:59:34
run
    AppRegistry.js:101:10
runApplication
    AppRegistry.js:195:26
__callFunction
    MessageQueue.js:349:47
<unknown>
    MessageQueue.js:106:26
__guard
    MessageQueue.js:297:10
callFunctionReturnFlushedQueue
    MessageQueue.js:105:17
```

matt-oakes, 2019-02-12:
Hi there,

This method is actually only available on iOS and from your screenshot, it looks like you are attempting to call it on Android. I recommend checking that you are on iOS before calling it (`Platform.OS === "ios"`).

What you likely want to do is to subscribe to location updates with a large distance filter so your callback only gets called when a large distance change happens. You can do this cross-platform like this:

```
RNLocation.configure({
    distanceFilter: 500, // Meters
    desiredAccuracy: {
      ios: "best",
      android: "balancedPowerAccuracy"
    }
})

// Subscribe
const unsubscribe = RNLocation.subscribeToLocationUpdates(locations => {
  ...
})

// Unsubscribe
unsubscribe();
```

I do think the error message for calling `startMonitoringSignificantLocationChanges` on Android could be improved, so I'll leave this open as a reminder for me to do that.

Let me know if you have any other issues.

arhmnsh, 2019-02-12:
Hi @matt-oakes,
Thanks for the reply! 
Got confused with `subscribeToLocationUpdates` and `subscribeToSignificantLocationUpdates`.

closed 2019-03-29 by matt-oakes