timfpark/react-native-location #43: subscribeToLocationUpdates() rarely returns new locations on Android

naftalibeder, 2019-02-14:
On Android, `subscribeToLocationUpdates()` returns a set of locations the first time it's called, but after that, it sends a new `locations` array extremely infrequently - on the order of once every 10 minutes (even if moving significantly).

On iOS, the method works as expected (using the same code below).

Relevant code is below. Are there any other settings I can adjust?

```
RNLocation.configure({
  distanceFilter: null, // have tried 0, 1, 10, 100
  desiredAccuracy: {
    ios: 'best',
    android: 'highAccuracy'
  }
});
```

```
unsubscribe = RNLocation.subscribeToLocationUpdates((locations) => {
  Alert.alert('locations', JSON.stringify(locations));
});
```

matt-oakes, 2019-02-15:
Hi @naftalibeder. Can you confirm what permissions you have added to your `AndroidManifest.xml` please?

naftalibeder, 2019-02-15:
@matt-oakes Sure! I’ve pasted the full permissions block from the manifest below. 

```
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE" />
```

matt-oakes, 2019-02-18:
Thanks! That all looks good.

Can you confirm that the app is running in the foreground? And also what version of Android are you seeing this on?

Android 7+ gets pretty restrictive on battery usage, so it's likely that you don't have the correct combination of settings to get such frequent updates. The details are here:

https://developer.android.com/guide/topics/location/battery

Looking at that, it looks like this library is missing some options for Android which are required to get frequent updates (frequency and max wait time). I will add those this week and release a new version of the library to see if that helps you.

jnystrom, 2019-02-21:
i am trying to use the library to get the compass direction of the device, can I use the `course` property for this?  I am seeing this same issue of not getting notified of location change.

naftalibeder, 2019-02-21:
@matt-oakes The app is indeed running in the foreground while the location update is failing to come through.

And yeah, I'm running Android version 8.0.0. I configured the library to run `setPriority(PRIORITY_HIGH_ACCURACY)`. But I look forward to the additional config fields - thanks for the reply!

matt-oakes, 2019-02-22:
Ok. It's on my list to add those new configuration properties, but I have some client work to be doing today. Keep an eye out here for any updates.

fuzunspm, 2019-02-23:
i also have the same issue. Even on example code it only updates once and stop. I also use react-native-maps to display map and it's updating fine with the same permission and project build

fuzunspm, 2019-02-23:
> @matt-oakes Sure! I’ve pasted the full permissions block from the manifest below.
> 
> ```
>     <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
> ```

is it ACCESS_COARSE_LOCATION or ACCESS_COURSE_LOCATION ? mine is set to latter

matt-oakes, 2019-02-23:
@fuzumspm It should be the one which I said (https://developer.android.com/reference/android/Manifest.permission.html#ACCESS_COARSE_LOCATION), however, if you want frequent precise location information then you need to ask for ACCESS_FINE_LOCATION as noted above.

matt-oakes, 2019-02-24:
I have just published [v2.2.0](https://github.com/timfpark/react-native-location/releases/tag/2.2.0) which includes three additional configuration options for Android which fixes the issues here. For example, to get a location update every 1 second you can call `configure` with these options:

```javascript
RNLocation.configure({
  desiredAccuracy: {
    ios: "best",
    android: "highAccuracy"
  },
  interval: 1000,
  maxWaitTime: 1000
});
```

Let me know if you have any problems with these.

fuzunspm, 2019-02-24:
Thank you, will post updates here

fuzunspm, 2019-02-25:
it's fixed the issue thank you

closed 2019-02-25 by matt-oakes

naftalibeder, 2019-02-25:
Works perfectly. Thank you so much for the wonderful library!

lyseiha, 2020-07-02:
Is it working in background or lock screen(android)?

abner01061998, 2020-12-18:
not working on foreground