timfpark/react-native-location #44: Native module cannot be null

karanpratapsingh, 2019-02-17:
Thanks for this library, it works good with android but with iOS it gives Native Module cannot be null error.

Things I did:

1. Installed the lib.
2. linked using react-native link
3. even linked location.coreframework lib

can someone help?

matt-oakes, 2019-02-18:
Hi @karanpratapsingh,

Can you confirm that you rebuilt the app after linking everything? Just refreshing the javascript will not be enough.

If you did, can you confirm that you are linking the correct Xcode target (if you have multiple of them) and that there are no warnings being produced about missing libraries when you build the app?

Thanks,
Matt

karanpratapsingh, 2019-02-18:
I followed the docs and rebuilt everything, great library, unfortunately
I’ve switched to navigator api as of now

matt-oakes, 2019-02-18:
Ok. Thanks for the report @karanpratapsingh. I'll confirm myself if this is working later and close this issue if it looks like it is. It sounds like something was "off" with the linking, however, if you've moved to another library then that's fine 👍

karanpratapsingh, 2019-02-18:
No problem bro, I’ll try it with a fresh project

closed 2019-02-19 by karanpratapsingh

anniewey, 2019-03-07:
@karanpratapsingh Restart packager and rerun your project will solve the issue

ankitjainOfficial, 2019-07-16:
@matt-oakes I had run into same problem. I unlinked the lib and tested, everything worked fine. Linked it back again, and the same error. Any suggestions, on what can be tried.

@karanpratapsingh Could you resolve the issue?

karanpratapsingh, 2019-07-17:
@ankitjainOfficial , I haven't tried this library since feb, as I switched to using navigator.getLocationAsync() that comes with react native and works good enough for simple stuff,

robertvorthman, 2019-07-17:
I fixed this issue by running ```pod install``` from the ios folder