timfpark/react-native-location #47: accuracy always 2000

denizbas92, 2019-03-01:
When I try to get my location ,the accuracy value that I obtained is always 2000, it never changed. Is there any bugs have to be fixed  or I am doing wrong somethink ?

matt-oakes, 2019-03-01:
Can you let me know:

* Which platform are you seeing this behaviour on?
* Is this on a real device or a simulator/emulator?
* What Android permissions do you have in `AndroidManifest.xml`?
* What permissions reasons do you have in `Info.plist`?
* What configuration options are you passing before subscribing to updates?

denizbas92, 2019-03-01:
=> My platform is Android.
 => I am testing it on a real device.
 => My permission in AndroidManifest.xml as shown below:
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>

=> These are my code : 
     componentWillMount() {
    RNLocation.configure({
      distanceFilter: 0, // Meters
      desiredAccuracy: {
        ios: "best",
        android: "balancedPowerAccuracy"
      },
      // Android only
      androidProvider: "auto",
      interval: 5000, // Milliseconds
      fastestInterval: 10000, // Milliseconds
      maxWaitTime: 5000, // Milliseconds
      // iOS Only
      activityType: "other",
      allowsBackgroundLocationUpdates: false,
      headingFilter: 1, // Degrees
      headingOrientation: "portrait",
      pausesLocationUpdatesAutomatically: false,
      showsBackgroundLocationIndicator: false,
    })
  }

  componentDidMount() {
    RNLocation.requestPermission({
      ios: "always",
      android: {
        detail: "coarse"
      }
    }).then(granted => {
      if (granted) {
        this.locationSubscription = RNLocation.subscribeToLocationUpdates(locations => {
          console.log(locations)
          /* Example location returned
          {
            speed: -1,
            longitude: -0.1337,
            latitude: 51.50998,
            accuracy: 5,
            heading: -1,
            altitude: 0,
            altitudeAccuracy: -1
            floor: 0
            timestamp: 1446007304457.029
          }
          */
        })
      }
    })
  }

=> The output is like this:
[ { timestamp: 1551433571853,
speedAccuracy: 0,
speed: 0,
 courseAccuracy: 0,
course: 0,
altitudeAccuracy: 0,
altitude: 0,
 accuracy: 2000,
longitude: 29.00299727646425,
latitude: 41.04504504504504 } ]

=> The devices I used are ;Android 6.0 and Android 7.0 (Real device)

matt-oakes, 2019-03-01:
The issue is that even though you have the correct permissions in `AndroidManifest.xml` for fine location, you are only requesting `coarse` permissions here:

```
RNLocation.requestPermission({
  ios: "always",
  android: {
    detail: "coarse"
  }
});
```

That should be changed to:

```
RNLocation.requestPermission({
  ios: "always",
  android: {
    detail: "fine"
  }
});
```

closed 2019-03-01 by matt-oakes

denizbas92, 2019-03-01:
Thank you very much for your answers,it worked