timfpark/react-native-location #48: Blue ticker IOS for location tracking remove/ stop location tracking based on premission.

harrie006, 2019-03-15:
I need to track location only when user allows "Always" permission for location tracking. The same functionality as Foursquare app.
The problem is as long as the permission for location tracking is always there is no ticker for location tracking in IOS. When I change permission to "when in use" from settings the ticker start showing. Here I want that the location tracking stops and blue ticker goes away

matt-oakes, 2019-03-15:
Sorry, can you explain what you mean by "ticker"? If possible, a reproducible demo would be very useful.

anhtuank7c, 2019-04-10:
May @harrie006  mention `blue bar` in https://developer.apple.com/library/archive/qa/qa1965/_index.html
Seem this module already support it: https://github.com/timfpark/react-native-location/blob/master/ios/RNLocation.m#L192
Note that we need `iOS 11+` to see that happen.

matt-oakes, 2019-04-10:
Ok. Thanks for letting me know. This should be possible with the options that the library currently has.

closed 2019-04-10 by matt-oakes