timfpark/react-native-location #51: How to open location service settings?

sinhpn92, 2019-04-06:
Hi,
I found the issue when my app is turn of location service in the settings panel. I have allowed location permission. However, I was turned off my device location service. So the location always null. 

How can i check the location settings and open settings panel to user turn on?

Thanks

matt-oakes, 2019-04-06:
👋 Checking the location services status is something we need to add. We’re tracking this in #19.

Keep an eye on progress there or submit a PR if you are able.

sinhpn92, 2019-04-08:
Thanks so much @matt-oakes 
I'm waiting for this... :)

bnash2501, 2019-07-22:
If you're looking to open up the iOS system preferences to enable services, it appears that post iOS10 that the API was turned private and it's usage runs the risk of getting banned.

https://forums.developer.apple.com/thread/108866

matt-oakes, 2019-07-24:
@bnash2501 Good to know. Thanks!