timfpark/react-native-location #52: Installation Bug on IOS CocoaPod

matamicen, 2019-04-06:
Hi,

I have this error: React/RCTBridge.h' file not found

it works on Android, I tried cocoapods without react-native link and with it, just in case, none of them works.

Any help?

Thanks.

matt-oakes, 2019-04-06:
👋 Have you gone into the `ios` Directory and run `pod install`? If so, can you send over he contents of your `Podfile` and `Podfile.lock`?

matamicen, 2019-04-06:
Hi Matt, 
Yes, I Did run pod install, then I did a new different project and did the same with the same results.

Here is my Podfile.lock

```
PODS:
  - React (0.11.0):
    - React/Core (= 0.11.0)
  - react-native-location (2.2.0):
    - React
  - React/Core (0.11.0)

DEPENDENCIES:
  - react-native-location (from `../node_modules/react-native-location/react-native-location.podspec`)

SPEC REPOS:
  https://github.com/cocoapods/specs.git:
    - React

EXTERNAL SOURCES:
  react-native-location:
    :path: "../node_modules/react-native-location/react-native-location.podspec"

SPEC CHECKSUMS:
  React: ab1a2e21deb34965c38328d5ec40cc7d12c6050a
  react-native-location: b7e0eacf93d4cfaf59fa65808987583a7fbe46cb

PODFILE CHECKSUM: 314f6a5677a6a7d77b2c6d435c4146e57e7e3c70

COCOAPODS: 1.6.1

```

Here is my Podfile:
```

# Uncomment the next line to define a global platform for your project
# platform :ios, '9.0'

target 'testLocation2' do
  # Uncomment the next line if you're using Swift or would like to use dynamic frameworks
  # use_frameworks!

  # Pods for testLocation2
  pod 'react-native-location', :path => '../node_modules/react-native-location/react-native-location.podspec'

  target 'testLocation2-tvOSTests' do
    inherit! :search_paths
    # Pods for testing
  end

  target 'testLocation2Tests' do
    inherit! :search_paths
    # Pods for testing
  end

end

target 'testLocation2-tvOS' do
  # Uncomment the next line if you're using Swift or would like to use dynamic frameworks
  # use_frameworks!

  # Pods for testLocation2-tvOS

  # target 'testLocation2-tvOSTests' do
  #   inherit! :search_paths
  #   # Pods for testing
  # end

end

```

Thanks for your help @matt-oakes  !!

matamicen, 2019-04-06:
This is my package.json:

```
{
  "name": "testLocation2",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest"
  },
  "dependencies": {
    "react": "16.8.3",
    "react-native": "0.59.3",
    "react-native-location": "^2.2.0"
  },
  "devDependencies": {
    "@babel/core": "7.4.3",
    "@babel/runtime": "7.4.3",
    "babel-jest": "24.7.1",
    "jest": "24.7.1",
    "metro-react-native-babel-preset": "0.53.1",
    "react-test-renderer": "16.8.3"
  },
  "jest": {
    "preset": "react-native"
  }
}

```

matt-oakes, 2019-04-06:
To use Cocoapods with any React Native library you need to move to having React Native itself as a Cocoapods dependency. The instructions for that are here:

https://facebook.github.io/react-native/docs/integration-with-existing-apps#configuring-cocoapods-dependencies

If you do not want to use Cocoapods, then you should delete your Podfile, Podfile.lock, and Pods directory and re-run the link command.

closed 2019-04-06 by matt-oakes

matamicen, 2019-04-06:
@matt-oakes thanks for your help, I know I am doing something wrong, but I can't detect what is it, for example this is What I did for another project where I added a Cocoapod library (Firebase) and it is working, so What Do I did different in this one that i Get the "React/RCTBridge.h' file not found" ? .... Thanks again.

Here are the files of another project with cocoapod:

this is my podfile.lock
```
PODS:
  - Firebase/AdMob (5.17.0):
    - Firebase/Core
    - Google-Mobile-Ads-SDK (~> 7.39)
  - Firebase/Core (5.17.0):
    - Firebase/CoreOnly
    - FirebaseAnalytics (= 5.6.0)
  - Firebase/CoreOnly (5.17.0):
    - FirebaseCore (= 5.3.0)
  - FirebaseAnalytics (5.6.0):
    - FirebaseCore (~> 5.3)
    - FirebaseInstanceID (~> 3.5)
    - GoogleAppMeasurement (= 5.6.0)
    - GoogleUtilities/AppDelegateSwizzler (~> 5.2)
    - GoogleUtilities/MethodSwizzler (~> 5.2)
    - GoogleUtilities/Network (~> 5.2)
    - "GoogleUtilities/NSData+zlib (~> 5.2)"
    - nanopb (~> 0.3)
  - FirebaseCore (5.3.0):
    - GoogleUtilities/Logger (~> 5.2)
  - FirebaseInstanceID (3.5.0):
    - FirebaseCore (~> 5.3)
    - GoogleUtilities/Environment (~> 5.3)
    - GoogleUtilities/UserDefaults (~> 5.3)
  - Google-Mobile-Ads-SDK (7.40.0)
  - GoogleAppMeasurement (5.6.0):
    - GoogleUtilities/AppDelegateSwizzler (~> 5.2)
    - GoogleUtilities/MethodSwizzler (~> 5.2)
    - GoogleUtilities/Network (~> 5.2)
    - "GoogleUtilities/NSData+zlib (~> 5.2)"
    - nanopb (~> 0.3)
  - GoogleUtilities/AppDelegateSwizzler (5.3.7):
    - GoogleUtilities/Environment
    - GoogleUtilities/Logger
    - GoogleUtilities/Network
  - GoogleUtilities/Environment (5.3.7)
  - GoogleUtilities/Logger (5.3.7):
    - GoogleUtilities/Environment
  - GoogleUtilities/MethodSwizzler (5.3.7):
    - GoogleUtilities/Logger
  - GoogleUtilities/Network (5.3.7):
    - GoogleUtilities/Logger
    - "GoogleUtilities/NSData+zlib"
    - GoogleUtilities/Reachability
  - "GoogleUtilities/NSData+zlib (5.3.7)"
  - GoogleUtilities/Reachability (5.3.7):
    - GoogleUtilities/Logger
  - GoogleUtilities/UserDefaults (5.3.7):
    - GoogleUtilities/Logger
  - nanopb (0.3.901):
    - nanopb/decode (= 0.3.901)
    - nanopb/encode (= 0.3.901)
  - nanopb/decode (0.3.901)
  - nanopb/encode (0.3.901)

DEPENDENCIES:
  - Firebase/AdMob
  - Firebase/Core

SPEC REPOS:
  https://github.com/cocoapods/specs.git:
    - Firebase
    - FirebaseAnalytics
    - FirebaseCore
    - FirebaseInstanceID
    - Google-Mobile-Ads-SDK
    - GoogleAppMeasurement
    - GoogleUtilities
    - nanopb

SPEC CHECKSUMS:
  Firebase: 59d557e064217fab6a03ff00baa73c06e73832e6
  FirebaseAnalytics: 75e4bbc6417d190cc98ec1f17c41a4fad4c2c976
  FirebaseCore: c0c4befb82374d6aef64d800e569f47625352edc
  FirebaseInstanceID: 4522aad88f69297622062c0e9ffccdee3dd9b151
  Google-Mobile-Ads-SDK: 9d1c38a83febea769470aa514a9c7954e2d1483d
  GoogleAppMeasurement: 008e04ecd8efedd97a693aea8634aefe220bd26e
  GoogleUtilities: 111a012f4c3a29c9e7c954c082fafd6ee3c999c0
  nanopb: 2901f78ea1b7b4015c860c2fdd1ea2fee1a18d48

PODFILE CHECKSUM: a53b2a207ebc35aa62259ef3ade5a8080e105121

COCOAPODS: 1.6.1
```

and this is mi podfile:
```

# Uncomment the next line to define a global platform for your project
# platform :ios, '9.0'

target 'sqsonativeV57' do
  # Uncomment the next line if you're using Swift or would like to use dynamic frameworks
  # use_frameworks!

  # Pods for sqsonativeV57
  pod 'Firebase/Core'
  pod 'Firebase/AdMob'

  target 'sqsonativeV57-tvOSTests' do
    inherit! :search_paths
    # Pods for testing
  end

  target 'sqsonativeV57Tests' do
    inherit! :search_paths
    # Pods for testing
  end

end

target 'sqsonativeV57-tvOS' do
  # Uncomment the next line if you're using Swift or would like to use dynamic frameworks
  # use_frameworks!

  # Pods for sqsonativeV57-tvOS

 # target 'sqsonativeV57-tvOSTests' do
 #   inherit! :search_paths
    # Pods for testing
#  end

end

```

matt-oakes, 2019-04-07:
The issue is that you need to link React Native itself as a Cocoapods dependency rather than the way it is currently set up (linked as a framework through Xcode). If you take a look at the link before and follow the instructions, it will guide you on how to do it.

Linking the Firebase dependencies is different as they don't know anything about React Native. If you link a React Native library through Cocoapods, you need to link React Native itself as a Cocoapod dependency.

matamicen, 2019-04-08:
@matt-oakes  @timfpark I got it .... I decided to avoid Cocoa, so with react native link works great in both Android and iOS, thanks buddy!