timfpark/react-native-location #56: I get sometimes error: "Attempt to invoke virtual method 'android.os.Looper"

roysG, 2019-04-16:
java.lang.NullPointerException: Attempt to invoke virtual method 'android.os.Looper android.app.Activity.getMainLooper()' on a null object reference
        at com.google.android.gms.common.api.GoogleApi.<init>(Unknown Source:50)
        at com.google.android.gms.location.FusedLocationProviderClient.<init>(Unknown Source:8)
        at com.google.android.gms.location.LocationServices.getFusedLocationProviderClient(Unknown Source:2)
        at com.github.reactnativecommunity.location.RNPlayServicesLocationProvider.<init>(RNPlayServicesLocationProvider.java:49)
        at com.github.reactnativecommunity.location.RNLocationModule.createPlayServicesLocationProvider(RNLocationModule.java:103)
        at com.github.reactnativecommunity.location.RNLocationModule.createDefaultLocationProvider(RNLocationModule.java:96)
        at com.github.reactnativecommunity.location.RNLocationModule.configure(RNLocationModule.java:39)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:158)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
        at android.os.Looper.loop(Looper.java:164)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:192)
        at java.lang.Thread.run(Thread.java:764)

And then the app is closed

matt-oakes, 2019-04-16:
Can you please send over:

* The code you are using to call the library
* The code you are using to link the library on Android (from `android/app/build.gradle`)
* The output of `./android/gradlew app:dependencies`

roysG, 2019-04-16:
codes:

1::
RNLocation.configure({
                    distanceFilter: 0, // Meters
                    desiredAccuracy: {
                        android: "highAccuracy",
                    },
                    headingFilter: 1, // Degrees
                    androidProvider: "auto",
                    interval: 10, // Milliseconds
                    fastestInterval: 10, // Milliseconds
                    maxWaitTime: 10, // Milliseconds
                });

RNLocation.requestPermission({
                    ios: "always",
                    android: {
                        detail: "fine",
                        rationale: {
                            title: "some title",
                            message: "some message",
                            buttonPositive: "ok",
                            buttonNegative: "no"
                        }
                    }
            }).then(granted => {
                    console.log('roy granted', granted);

                    if (granted) {
                        console.log('roy enabled watchPosition');

                        this.unsubscribe = RNLocation.subscribeToLocationUpdates(async locations => {
                            console.log('get location');
                        });
                    }
                });

2:::

apply plugin: "com.android.application"
import com.android.build.OutputFile

/**
 * The react.gradle file registers a task for each build variant (e.g. bundleDebugJsAndAssets
 * and bundleReleaseJsAndAssets).
 * These basically call `react-native bundle` with the correct arguments during the Android build
 * cycle. By default, bundleDebugJsAndAssets is skipped, as in debug/dev mode we prefer to load the
 * bundle directly from the development server. Below you can see all the possible configurations
 * and their defaults. If you decide to add a configuration block, make sure to add it before the
 * `apply from: "../../node_modules/react-native/react.gradle"` line.
 *
 * project.ext.react = [
 *   // the name of the generated asset file containing your JS bundle
 *   bundleAssetName: "index.android.bundle",
 *
 *   // the entry file for bundle generation
 *   entryFile: "index.android.js",
 *
 *   // whether to bundle JS and assets in debug mode
 *   bundleInDebug: false,
 *
 *   // whether to bundle JS and assets in release mode
 *   bundleInRelease: true,
 *
 *   // whether to bundle JS and assets in another build variant (if configured).
 *   // See http://tools.android.com/tech-docs/new-build-system/user-guide#TOC-Build-Variants
 *   // The configuration property can be in the following formats
 *   //         'bundleIn${productFlavor}${buildType}'
 *   //         'bundleIn${buildType}'
[... 144 lines cut ...]

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
task copyDownloadableDepsToLibs(type: Copy) {
    from configurations.compile
    into 'libs'
}

3::

 bash: ./android/gradlew app:dependencies

FAILURE: Build failed with an exception.

* What went wrong:
Project 'app' not found in root project 'app'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 0s

Let me know if you need more details.

matt-oakes, 2019-04-16:
Sorry, can you actually run `./gradlew app:dependencies` from inside the `android` directory. I thought it would work with the prefix.

Everything you've sent before looks ok. Some more details which might be useful:

* Are you able to provide a simple reproduction repository which shows the issue?
* What are you running this on?
* An actual device or emulator?
* What Android version?
* Does it have Google Play Services installed?

Thanks!

tkuenzle, 2019-05-24:
Thanks a lot for this nice library, it comes in really handy! We experience the same issue on production (we get crash reports from firebase). So far it affects around 100 Users (out of 13000) and it seems to happen most of the time when the app is in background (95%). 

It happens on all Android versions, that we support (5-9).

Our dependency tree is the following:

```
prodReleaseCompileClasspath - Resolved configuration for compilation for variant: prodRelease
+--- org.webkit:android-jsc-intl:r241213
+--- project :react-native-gesture-handler
+--- project :react-native-firebase
|    +--- com.facebook.react:react-native:+ -> 0.59.3
|    |    +--- com.facebook.infer.annotation:infer-annotation:0.11.2
|    |    |    \--- com.google.code.findbugs:jsr305:3.0.1 -> 3.0.2
|    |    +--- javax.inject:javax.inject:1
|    |    +--- com.android.support:appcompat-v7:28.0.0
|    |    |    +--- com.android.support:support-annotations:28.0.0
|    |    |    +--- com.android.support:support-compat:28.0.0
|    |    |    |    +--- com.android.support:support-annotations:28.0.0
|    |    |    |    +--- com.android.support:collections:28.0.0
|    |    |    |    |    \--- com.android.support:support-annotations:28.0.0
|    |    |    |    +--- android.arch.lifecycle:runtime:1.1.1
|    |    |    |    |    +--- android.arch.lifecycle:common:1.1.1
|    |    |    |    |    |    \--- com.android.support:support-annotations:26.1.0 -> 28.0.0
|    |    |    |    |    +--- android.arch.core:common:1.1.1
|    |    |    |    |    |    \--- com.android.support:support-annotations:26.1.0 -> 28.0.0
|    |    |    |    |    \--- com.android.support:support-annotations:26.1.0 -> 28.0.0
|    |    |    |    \--- com.android.support:versionedparcelable:28.0.0
|    |    |    |         +--- com.android.support:support-annotations:28.0.0
|    |    |    |         \--- com.android.support:collections:28.0.0 (*)
|    |    |    +--- com.android.support:collections:28.0.0 (*)
|    |    |    +--- com.android.support:cursoradapter:28.0.0
|    |    |    |    \--- com.android.support:support-annotations:28.0.0
|    |    |    +--- com.android.support:support-core-utils:28.0.0
|    |    |    |    +--- com.android.support:support-annotations:28.0.0
|    |    |    |    +--- com.android.support:support-compat:28.0.0 (*)
|    |    |    |    +--- com.android.support:documentfile:28.0.0
|    |    |    |    |    \--- com.android.support:support-annotations:28.0.0
|    |    |    |    +--- com.android.support:loader:28.0.0
|    |    |    |    |    +--- com.android.support:support-annotations:28.0.0
|    |    |    |    |    +--- com.android.support:support-compat:28.0.0 (*)
|    |    |    |    |    +--- android.arch.lifecycle:livedata:1.1.1
|    |    |    |    |    |    +--- android.arch.core:runtime:1.1.1
|    |    |    |    |    |    |    +--- com.android.support:support-annotations:26.1.0 -> 28.0.0
|    |    |    |    |    |    |    \--- android.arch.core:common:1.1.1 (*)
|    |    |    |    |    |    +--- android.arch.lifecycle:livedata-core:1.1.1
|    |    |    |    |    |    |    +--- android.arch.lifecycle:common:1.1.1 (*)
|    |    |    |    |    |    |    +--- android.arch.core:common:1.1.1 (*)
|    |    |    |    |    |    |    \--- android.arch.core:runtime:1.1.1 (*)
|    |    |    |    |    |    \--- android.arch.core:common:1.1.1 (*)
|    |    |    |    |    \--- android.arch.lifecycle:viewmodel:1.1.1
|    |    |    |    |         \--- com.android.support:support-annotations:26.1.0 -> 28.0.0
|    |    |    |    +--- com.android.support:localbroadcastmanager:28.0.0
|    |    |    |    |    \--- com.android.support:support-annotations:28.0.0
|    |    |    |    \--- com.android.support:print:28.0.0
|    |    |    |         \--- com.android.support:support-annotations:28.0.0
|    |    |    +--- com.android.support:support-fragment:28.0.0
|    |    |    |    +--- com.android.support:support-compat:28.0.0 (*)
|    |    |    |    +--- com.android.support:support-core-ui:28.0.0
|    |    |    |    |    +--- com.android.support:support-annotations:28.0.0
|    |    |    |    |    +--- com.android.support:support-compat:28.0.0 (*)
|    |    |    |    |    +--- com.android.support:support-core-utils:28.0.0 (*)
|    |    |    |    |    +--- com.android.support:customview:28.0.0
|    |    |    |    |    |    +--- com.android.support:support-annotations:28.0.0
|    |    |    |    |    |    \--- com.android.support:support-compat:28.0.0 (*)
|    |    |    |    |    +--- com.android.support:viewpager:28.0.0
|    |    |    |    |    |    +--- com.android.support:support-annotations:28.0.0
|    |    |    |    |    |    +--- com.android.support:support-compat:28.0.0 (*)
|    |    |    |    |    |    \--- com.android.support:customview:28.0.0 (*)
|    |    |    |    |    +--- com.android.support:coordinatorlayout:28.0.0
[... 355 lines cut ...]
+--- com.google.android.gms:play-services-measurement-sdk-api:16.4.0 (*)
+--- com.google.firebase:firebase-measurement-connector:17.0.1 (*)
+--- com.google.android.gms:play-services-measurement-sdk:16.4.0 (*)
+--- com.google.firebase:firebase-messaging:17.5.0 (*)
+--- com.google.firebase:firebase-config:16.5.0 (*)
+--- com.google.firebase:firebase-abt:17.1.0 (*)
+--- com.google.firebase:firebase-invites:16.1.1 (*)
+--- com.google.android.gms:play-services-appinvite:16.1.1 (*)
+--- com.google.firebase:firebase-dynamic-links:16.1.8 (*)
+--- com.google.firebase:firebase-perf:16.2.5 (*)
+--- me.leolin:ShortcutBadger:1.1.21
[... cut ...]
+--- io.fabric.sdk.android:fabric:1.4.4
+--- com.crashlytics.sdk.android:crashlytics-core:2.6.4 (*)
+--- com.crashlytics.sdk.android:beta:1.2.10 (*)
+--- com.google.android.gms:play-services-places:16.0.0 (*)
+--- com.google.android.gms:play-services-vision:17.0.2 (*)
+--- com.google.android.gms:play-services-vision-common:17.0.2 (*)
\--- com.google.android.gms:play-services-flags:16.0.1 (*)

(*) - dependencies omitted (listed previously)
```

Unfortunately it does not happen always and I was not able to reproduce it locally yet.

closed 2019-06-02 by matt-oakes

matt-oakes, 2019-06-03:
Fixed in 2.2.1: https://github.com/timfpark/react-native-location/releases/tag/2.2.1

PR #65 (2019-06-02): Use context for location provider if activity is not available
The current activity may be null when the location provider is created, in which case we can use the React application context instead. Alternatively we could wait for the activity to become non-null, but that would add some more complexity.

Resolves: #56