timfpark/react-native-location #57: Possible Unhandled Promise Rejection

HPieters, 2019-04-17:
Trying to solve an issue I am having with this otherwise great library! On Android I am getting the following warning:

```
Possible Unhandled Promise Rejection (id: 0) 
Error: Error configuring react-native-location
```

This occurs on my Android device after I have agreed to all permissions, but turn off my location services / GPS. I can check if GPS is enabled with another [library](https://github.com/neuberoliveira/react-native-gps-state), however I do like the in-app pop up asking the user to turn GPS back on. This dialog is triggered when I run `RNLocation.configure` however if you cancel / decline the dialog `RNLocation.configure` triggered the above warning.

matt-oakes, 2019-04-17:
This error is because you are not handling the promise rejection. You need to either add a `catch` to your promise chain or surround your code in ` try`/`catch` is you are using `async`/`await`.

HPieters, 2019-04-18:
@matt-oakes I tried adding a try catch around the code but that doesn't help. But imo RNLocation.configure is a function not a promise so that makes sense? 

This is the code:
```
    RNLocation.configure({
      androidProvider: 'auto',
      desiredAccuracy: {
        android: 'highAccuracy',
        ios: 'best',
      },
      distanceFilter: 0,
      maxWaitTime: 10000,
    })
``` 

If I don't add this code I have no warning, If I add this and I play out the scenario as described I do get the warning. 

I will try to make a reproduction so you can see. I think it has to do with the popup being triggered (asking for GPS settings to be turned on) by the configure function and that if I decline GPS nothing is there to handle the rejection by the user?

matt-oakes, 2019-04-18:
Ah! I see the issue. The native code is returning a `Promise`, but the libraries JS code is not returning it for you to handle.

tkuenzle, 2019-06-18:
@matt-oakes I just dropped a PR for this, let me know if there is something I need to change.

closed 2019-06-26 by matt-oakes

viv3kk, 2019-09-28:
still getting this error on 2.3.0 ...
On which version it is fixed ??

alextorn, 2019-11-06:
Now promise returned only in case when you accept the dialog. 

> However if you cancel / decline the dialog RNLocation.configure triggers the above warning.
still exists

viv3kk, 2019-12-20:
how to handle this scenario

noway, 2019-12-31:
@viv3kk 

> how to handle this scenario

```javascript
    try {
      await RNLocation.configure({
        distanceFilter: 0.0, // meters
        desiredAccuracy: {
          ios: 'bestForNavigation',
          android: 'highAccuracy',
        },
        ...
     })
    }
    catch (e) {
      Alert.alert('Location Services', 'Please enable location services')
      // todo: set redux state for disabled state, navigate to disabled state screen etc
    }
```