timfpark/react-native-location #59: subscribeToPermissionUpdates not working on Android

reinvanimschoot, 2019-04-26:
Hi guys

First of all, thanks for this amazing library! It really helped us tackle a number of difficult problems.

One thing, however, that I don't seem to get working is, `subscribeToPermissionUpdates` on Android. It seems as though it is never being called. 

Beyond that, the library works great on Android and I am pretty sure I installed it correctly.

I am using:

react-native 0.59.5
react-native-location: 2.2.0

matt-oakes, 2019-05-26:
Apologies for the delay in responding to this.

Can you provide some more details? I'm not sure what is "never being called". Do you mean that you are never receiving location updates?

If so, some possible things to check are:

* Do you have the correct permissions?
* Have you called `configure` with the correct settings?

Elvinra, 2019-06-14:
I think this is because it's not implemented on android 

https://github.com/timfpark/react-native-location/search?q=authorizationStatusDidChange&unscoped_q=authorizationStatusDidChange

shshpnv, 2019-07-17:
> Apologies for the delay in responding to this.
>
> Can you provide some more details? I'm not sure what is "never being called". Do you mean that you are never receiving location updates?
>
> If so, some [...]

he said PermissionUpdate , not Location

Under-Warz, 2019-10-02:
+1 for this feature on Android

Under-Warz, 2019-10-02:
@reinvanimschoot waiting for the PR, you can check the current permission on app resume in you app.tsx by exemple :

```
handleAppStateChange = async (nextAppState: AppStateStatus) => {
    ...

    // Check if permission changed
    if (Platform.OS === "android") {
      const granted = await LocationManager.checkPermission()
      if (granted) {
        this.subscribeLocationUpdates()
      }
    }

    this.setState({ appState: nextAppState })
  }
```

sunilloharshoreline1, 2021-10-17:
@tdranv this could be possible running background Service in native Android (java/kotlin), one possible way to handle in react-native would be to check the location after some time e.g. 1 second using setInterval method, but this would be the workaround and wouldn't be the ideal solution I guess.