timfpark/react-native-location #60: subscribeToLocationUpdates  gets no value in android,I can't get the value, please help me

zhaochu0821, 2019-04-27:
[image]
[image]
[image]
[image]

zhaochu0821, 2019-04-27:
@matt-oakes

matt-oakes, 2019-05-08:
Hello 👋 Thanks for submitting this issue. To help us triage your issue, can you please resubmit this issue using the templates provided.

closed 2019-05-08 by matt-oakes

hieuplasma, 2021-10-06:
I'm facing this problem right now, did you have solution for this? Help me pls!