timfpark/react-native-location #63: Build tools error, cannot find symbol

shubhang93, 2019-05-21:
cannot find symbol
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                                                        ^
  symbol:   variable O
  location: class VERSION_CODES

I guess the build tools version is old

matt-oakes, 2019-05-26:
👋 Have you changed your Android SDK versions at all? Can you post the contents of your `android/build.gradle` file?

ethanysc, 2019-05-28:
@shubhang93 we set the versions in `react-native-location/android/build.gradle` to ours and the build error went away 

```
def DEFAULT_COMPILE_SDK_VERSION          = 26
def DEFAULT_BUILD_TOOLS_VERSION          = "26.0.1"
def DEFAULT_TARGET_SDK_VERSION           = 26
def DEFAULT_GOOGLE_PLAY_SERVICES_VERSION = "16.+"
```

@matt-oakes if a PR can be made soon that'd be great, or else we can fork as well. Thank you!

matt-oakes, 2019-06-10:
Closing. See comment in #67.

closed 2019-06-10 by matt-oakes

theRealSheng, 2020-05-31:
I'm getting the same error as this.

    compileSdkVersion 28
    minSdkVersion 23
    targetSdkVersion 28

Can you please advise how I can solve this issue? Thanks in advance!