timfpark/react-native-location #66: Request location manually

radeno, 2019-05-28:
Hello,

I like this project and we are using it without any issues.

Currently, we are heading a situation where we don't want to have an automatically updated location every time it is changed. 
But we want to make it manually on specific situations (screen loaded, etc) because every time is location updated and stored into redux, some "ugly" rerendering is made. We should increase `distanceFilter` or compare values in component lifetime cycle method. But it is so clean as should be.

In this package core is some undocumented methods to automatically starting and stopping location updates.

Maybe `startUpdatingLocation` and `stopUpdatingLocation` should be documented. But still, need to be implemented a trigger to get location manually. And after all, option to enable or disable automatic updating location when RNLocation is configured. 

iOS: https://developer.apple.com/documentation/corelocation/cllocationmanager/1620548-requestlocation?language=objc
Android: https://developer.android.com/reference/android/location/LocationManager.html#requestSingleUpdate(java.lang.String,%2520android.location.LocationListener,%2520android.os.Looper)

Take this issue just like a discussion for the feature request.

matt-oakes, 2019-05-28:
:wave: Thanks for the request!

You should already be able to do this by adding a listener and unsubscribing when you no longer want updates.

The methods mentioned are internal implementation details and should not be used.

closed 2019-05-28 by matt-oakes

radeno, 2019-05-28:
You are right about subscribing and unaubscribing. But what about If i want to have one global subscriber and firing location request when i need to ?

Subscribing and unsubscribing and again and again for example 10 screens looks very complicated solution.

But have one app/root subscriber connected to Redux and just updating when is need looks more easier.

For example:
- for some screens i need position no older than 10 seconds. Fire request
- for some screens is 1 minúte enought. Do not fire request
- same for accuracy
- speed, dIstance and any other measure

If i need to making new listener for every screen on and off. It ist little bit “scratching with left hand on right ear”. Even when i need just one manual request to update my location

radeno, 2019-05-28:
Maybe any other use cases.

Request location only once on rendering screen. But have one global listener as saved whole object to Redux or React Context. And based on it, when location is requested, tadaaa, rerender with fresh data.

If people are in move, by walk, bike, in car, in train ,in bus, in any other vehicle. And your list of points if sorted by distance, you want to render it only once. Instead you get very nice flickering effect because rerendering :)

Handling this use cases with every new listener or comparing data in "should render" method is not clean and its bloat code a bit.