timfpark/react-native-location #71: Can't get the location from 'getLatestLocation'

mingxin-yang, 2019-06-25:
This is my code to get the last location:
```js
RNLocation.requestPermission({
      ios: "whenInUse",
      android: {
        detail: "coarse"
      },
    }).then(granted => {
      if (granted) {
        RNLocation.configure({
          distanceFilter: 0, // Meters
        });
        RNLocation.getLatestLocation({ timeout: 60000 }).then(locations => {
          if (locations != null) {
            global.longitude = locations.longitude;
            global.latitude = locations.latitude;
            callback(locations);
          }
        });
      }
    });
```
the locations is null,
also  I want to know that how to request the Permission('coarse' and 'fine') simultaneously

rafalzawadzki, 2019-06-25:
I have the same issue. What would be the recommended course of action in case the location is null? How can we prevent this?

matt-oakes, 2019-07-01:
If the location is `null` then it means that the device does not currently have a location which meets the settings and permissions you have. This is a prettily valid state to be in, and you need to handle the case where permissions are correct, but there is no location. You could try changing your configuration to give you a bigger chance of being returning a location.

I'm going to close this issue, but please open a new one if you think that there is a bug with the library.

robertvorthman, 2019-08-19:
I've found that `getLatestLocation` always returns null if there is an active subscription with `subscribeToLocationUpdates`

matt-oakes, 2019-08-22:
@robertvorthman Interesting... I'll reopen this to take a look at that.

prasanthyejje, 2019-10-15:
> getLatestLocation

so in that case do i need to remove the subscribeToLocationUpdates ?

rahamin1, 2019-12-11:
> If the location is null then it means that the device does not currently have a location which meets the settings and permissions you have. This is a prettily valid state to be in, and you need to handle the case where permissions are correct, but there is no location. 

Can you list some examples when this happens, assuming that I am configuring with `distanceFilter: 0, desiredAccuracy: 'highAccuracy'` ?

mingxin-yang, 2020-01-08:
I tried that  It is work when I use `detail: "fine"` in HuaWei, but not in Samsung or OPPO, use `detail: "coarse"` is fine in Samsung or OPPO, so I think if we can use the both ?

ckasek, 2020-02-26:
> @robertvorthman Interesting... I'll reopen this to take a look at that.

@matt-oakes Did you ever take a look at this? I'm noticing the same thing - If I'm subscribed to location updates, then calling getLatestLocation always returns null. Even if I unsubscribe, it still returns null.

ollydixon, 2020-04-19:
Same on iOS, location is so broken on React-native.

ozturkilgim, 2020-05-18:
Same Issue

ZaidSA, 2020-11-05:
Guys any updates on this? @matt-oakes

paddy57, 2021-04-24:
facing same issue on iOS, any updates ?

KrifaYounes, 2021-06-09:
same issue on iOS
any updates ?
enable to use getLatestLocation with RNLocation.subscribeToLocationUpdates

Gopinath-dartexon, 2023-01-03:
Facing this issue with Android 12