timfpark/react-native-location #72: Task :react-native-location:compileDebugJavaWithJavac FAILED

alittletf, 2019-06-25:
Hi all, 

I followed the installation instructions for Android and I seem to be getting this error when trying to build the app. Can't seem to find much on google so here I am...

RN: 0.58.6
react-native-location: 2.2.1

I used React Native Link but to install but then verified it all worked through manual installation. 

```
> Task :react-native-location:compileDebugJavaWithJavac FAILED
/Users/me/myapp/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:34: error: cannot find symbol
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                                                        ^
  symbol:   variable O
  location: class VERSION_CODES
/Users/me/myapp/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:35: error: cannot find symbol
            map.putDouble("altitudeAccuracy", location.getVerticalAccuracyMeters());
                                                      ^
  symbol:   method getVerticalAccuracyMeters()
  location: variable location of type Location
/Users/me/myapp/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:40: error: cannot find symbol
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                                                        ^
  symbol:   variable O
  location: class VERSION_CODES
/Users/me/myapp/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:41: error: cannot find symbol
            map.putDouble("courseAccuracy", location.getBearingAccuracyDegrees());
                                                    ^
  symbol:   method getBearingAccuracyDegrees()
  location: variable location of type Location
/Users/me/myapp/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:46: error: cannot find symbol
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
                                                        ^
  symbol:   variable O
  location: class VERSION_CODES
/Users/me/myapp/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:47: error: cannot find symbol
            map.putDouble("speedAccuracy", location.getSpeedAccuracyMetersPerSecond());
                                                   ^
  symbol:   method getSpeedAccuracyMetersPerSecond()
  location: variable location of type Location
6 errors
```

thank you in advance!

alittletf, 2019-06-26:
I realized that I was missing come code in my android/build.gradle. It seemed I had some version miss matches. I updated it to: 

```
    ext {

        compileSdkVersion = 28
        targetSdkVersion = 28
        buildToolsVersion = "28.0.3"
        supportLibVersion = "28.0.0"
        googlePlayServicesAuthVersion = "16.0.0"
    }
```

and everything worked perfect.

closed 2019-06-26 by alittletf

dinithminura, 2019-07-14:
Facing the same issue, my android/build.gradle file same as the @alittletf. 

System:
    OS: macOS 10.14.5
    CPU: (8) x64 Intel(R) Core(TM) i7-4770HQ CPU @ 2.20GHz
    Memory: 19.38 MB / 16.00 GB
    Shell: 3.2.57 - /bin/bash
  Binaries:
    Node: 12.4.0 - /usr/local/bin/node
    Yarn: 1.17.0 - /usr/local/bin/yarn
    npm: 6.9.0 - /usr/local/bin/npm
    Watchman: 4.9.0 - /usr/local/bin/watchman
  SDKs:
    iOS SDK:
      Platforms: iOS 12.2, macOS 10.14, tvOS 12.2, watchOS 5.2
    Android SDK:
      API Levels: 28
      Build Tools: 28.0.3
  IDEs:
    Android Studio: 3.4 AI-183.6156.11.34.5522156
    Xcode: 10.2.1/10E1001 - /usr/bin/xcodebuild
  npmPackages:
    react: 16.8.6 => 16.8.6 
    react-native: 0.60.3 => 0.60.3 
  npmGlobalPackages:
    react-native-cli: 2.0.1

amirmatin, 2019-07-31:
I am still having this issue. after applying @alittletf  fix, the app crashes right at the beginning.