timfpark/react-native-location #74: android 6.0 crash

kingctan, 2019-07-08:
```
ReactNative: Exception in native call
    java.lang.RuntimeException: Could not invoke RNLocation.startUpdatingLocation
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:383)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:158)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:743)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29)
        at android.os.Looper.loop(Looper.java:150)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:232)
        at java.lang.Thread.run(Thread.java:833)
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Method.invoke(Native Method)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:158) 
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method) 
        at android.os.Handler.handleCallback(Handler.java:743) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29) 
        at android.os.Looper.loop(Looper.java:150) 
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:232) 
        at java.lang.Thread.run(Thread.java:833) 
     Caused by: java.lang.VirtualMachineError: Invoking <init> with bad arg 0, type 'Ljava/lang/String;' not instance of 'Lcom/facebook/react/animation/Animation;'
        at com.github.reactnativecommunity.location.Utils.emitEvent(Utils.java:24)
        at com.github.reactnativecommunity.location.RNStandardLocationProvider.processLocation(RNStandardLocationProvider.java:130)
        at com.github.reactnativecommunity.location.RNStandardLocationProvider.setupListening(RNStandardLocationProvider.java:100)
        at com.github.reactnativecommunity.location.RNStandardLocationProvider.startUpdatingLocation(RNStandardLocationProvider.java:40)
        at com.github.reactnativecommunity.location.RNLocationModule.startUpdatingLocation(RNLocationModule.java:68)
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372) 
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:158) 
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method) 
        at android.os.Handler.handleCallback(Handler.java:743) 
        at android.os.Handler.dispatchMessage(Handler.java:95) 
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:29) 
        at android.os.Looper.loop(Looper.java:150) 
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:232) 
        at java.lang.Thread.run(Thread.java:833) 
```

MCherrey, 2020-04-24:
I made a pull request that should fix this. It is due to the component using the Support libraries instead of Androidx. #123 

I know I am kinda late on this for you, but hopefully this gets bumped to the maintainer's attention.

Haripriya-dhall, 2024-09-03:
I am getting similar issue, can anyone help me please

Sonu-dhall, 2024-09-03:
I am getting same error can anyone help me. How to solve this error.
ava.lang.RuntimeException: Could not invoke RNLocation.startUpdatingLocation
                                                                                                    	at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:383)
                                                                                                    	at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:188)
                                                                                                    	at com.facebook.jni.NativeRunnable.run(Native Method)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:959)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:100)
                                                                                                    	at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:232)
                                                                                                    	at android.os.Looper.loop(Looper.java:317)
                                                                                                    	at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:228)
                                                                                                    	at java.lang.Thread.run(Thread.java:1012)
                                                                                                    Caused by: java.lang.reflect.InvocationTargetException
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
                                                                                                    	... 9 more
                                                                                                    Caused by: java.lang.IncompatibleClassChangeError: Found interface com.google.android.gms.location.FusedLocationProviderClient, but class was expected (declaration of 'com.google.android.gms.location.FusedLocationProviderClient' appears in /data/app/~~7smJo3I08Cqw-Y36dUp1nA==/com.propstream-jtMm5nAAUg-zx1zDRWz1Sg==/base.apk!classes19.dex)
                                                                                                    	at com.github.reactnativecommunity.location.RNPlayServicesLocationProvider.reSetUpLocationListeners(RNPlayServicesLocationProvider.java:233)
                                                                                                    	at com.github.reactnativecommunity.location.RNPlayServicesLocationProvider.startUpdatingLocation(RNPlayServicesLocationProvider.java:190)
                                                                                                    	at com.github.reactnativecommunity.location.RNLocationModule.startUpdatingLocation(RNLocationModule.java:71)
                                                                                                    	... 11 more