timfpark/react-native-location #78: Error: Could not find the RNLocation native module

pierroo, 2019-07-31:
Hi,

I installed the latest version of this module.

But when I build and launch on a simulator, I get a big red error saying: 
"Native module cannot be null"

In my debugger it says: Could not find the RNLocation native module

I have used the Cocoapods installation method, so I don't see what I might have missed.

Please let me know how I could fix this issue, 
Thank you!

dbauducco, 2019-09-05:
+1

Exact same issue! Any solutions?

I can provide more details if needed.

iwakoscott, 2019-12-17:
+ 1

I am having the same issues when trying to import `react-native-location` from inside my `jest` test. I also get a `console.warn` that says - 

```bash
Could not find the RNLocation native module. Have you correctly linked react-native-location and rebuilt your app?
```

but I followed the necessary steps to `link` for `ios` and `android`...

iwakoscott, 2019-12-17:
So, I was able to get my `jest` tests to run again. It turns out I need to mock out `react-native-location` in my `setup.js` file like so,

```javascript
jest.mock('react-native-location', () => ({
  __esModule: true,
  default: {
    requestPermission: jest.fn(() => Promise.resolve(true)),
    getLatestLocation: jest.fn(() => Promise.resolve()),
  },
}));
```

I hope this helps.

MCherrey, 2020-04-24:
I have a patch here that fixes it. Would love it if the maintainer pulled it.