timfpark/react-native-location #82: Fatal Exception: java.lang.NullPointerException (v2.3.0)

viv3kk, 2019-08-13:
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.app.Activity.startIntentSenderForResult(android.content.IntentSender, int, android.content.Intent, int, int, int)' on a null object reference
       at com.google.android.gms.common.api.Status.startResolutionForResult + 17(:17)
       at com.google.android.gms.common.api.ResolvableApiException.startResolutionForResult + 3(:3)
       at com.github.reactnativecommunity.location.RNPlayServicesLocationProvider$2.onFailure + 175(RNPlayServicesLocationProvider.java:175)
       at com.google.android.gms.tasks.zzl.run + 4(:4)
       at android.os.Handler.handleCallback + 739(Handler.java:739)
       at android.os.Handler.dispatchMessage + 95(Handler.java:95)
       at android.os.Looper.loop + 179(Looper.java:179)
       at android.app.ActivityThread.main + 5730(ActivityThread.java:5730)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run + 791(ZygoteInit.java:791)
       at com.android.internal.os.ZygoteInit.main + 681(ZygoteInit.java:681)

matt-oakes, 2019-08-17:
👋 Can you confirm which version of the library and React Native you are using?

mika-lindell, 2019-10-04:
I'm getting this same error:

* RN 0.59.10
* react-native-location: 2.5.0

* Play services: 19.2.75
* Android 8.10

To reproduce:
* Call RNLocation.setConfig({androidProvider: 'auto'}) during app startup
* Set in Android settings:  Location & Security --> Privacy --> Location Mode --> Device only
* Then restart the app

One time I got it to show the play services dialog about changing your location settings, but all the other times the app will crash.

* Calling RNLocation.setConfig({androidProvider: 'standard'}) will not cause a crash. 
* Not calling setConfig during app start will not cause a crash. I see the library tries to set play services as default provider, so that's interesting.

mika-lindell, 2019-10-30:
This also happens on android devices with location turned off on device level.

For a workaround we check if network provider is on and location is available on device. If not, we force the standard provider

mika-lindell, 2019-11-15:
We were able to find the true cause and remove the workarounds. 

There is moment in our app when one activity is opened and another closed.
Around this time we will also start our location manager. Sometimes the location manager would start before the old activity was closed, and get the context of that activity from react native. Then react-native-location would crash when it tried to open the play services -dialog in the context of the closed activity (the one asking user to turn on high accuracy positioning).

TL:DR; Was because of closing and opening activity and wrong time, our app was to blame.

noway, 2020-01-07:
experiencing this too

360cid, 2020-01-15:
I am seeing this crash on an app that uses [CodePush](https://github.com/microsoft/react-native-code-push) to apply over-the-air updates, and the behavior we're experiencing is consistent with what @mika-lindell reported.

In our case we have a number of screens where we call `RNLocation.configure()` with different accuracy settings. If a user has Google Play Location Services disabled in device settings, the Play Services dialog will display every time `RNLocation.configure()` is called. If a CodePush restart is also triggered around this time, the app will crash. 

We are currently working around it by wrapping all calls to `RNLocation.configure` with `codepush.disallowRestart()`. This seems to be helping, but hasn't totally fixed the crash as there doesn't appear to be a way to wait to reallow the restart until after the Play Services dialog has been dismissed.

For reference we're on RN 0.60.6, react-native-location 2.2.1, Play Services 17.0.0.

pedrolopes10, 2020-06-04:
Hi there! I have crashlytics displaying a lots of NullPointerException
com.github.reactnativecommunity.location.RNPlayServicesLocationProvider$2.onFailure (RNPlayServicesLocationProvider.java:175).

They all are from 100% background state. 

I do run RNLocation.checkPermission({ios: 'always', android: { detail: 'fine' }}) on a backgroundTask.

Help please

pedrolopes10, 2020-06-06:
Well, activity != null solved the issue. I don't know the impact that may have to do a pr.

ali-irawan, 2021-01-16:
I also have this crash consistenly
react-native-location: 2.5.0

the androidProvider in type.td.ts is "auto" | "fused" | "built-in"
https://www.npmjs.com/package/react-native-location/v/2.5.0

but using "built-in" is not working, i need to use "standard"

then I can't find any source 2.5.0 in the github, but do find the 2.5.0 version in npmjs.com

the crash issue something in here

```
// On failure, attempt to recover. If we can't then reject the promise
        task.addOnFailureListener(new OnFailureListener() {
            @Override
            public void onFailure(@NonNull Exception e) {
                if (e instanceof ResolvableApiException) {
                    // Location settings are not satisfied, but this can be fixed
                    // by showing the user a dialog.
                    try {
                        // Save the settings we're currently trying to use along with the promise
                        pendingConfigureActivity = new WeakReference<>(activity);
                        pendingConfigureOptions = options;
                        pendingConfigurePromise = promise;
                        // Show the dialog by calling startResolutionForResult(),
                        // and check the result in onActivityResult().
                        ResolvableApiException resolvable = (ResolvableApiException) e;
                        resolvable.startResolutionForResult(activity, REQUEST_CHECK_SETTINGS);
                    } catch (IntentSender.SendIntentException sendEx) {
                        // Ignore the error.
                    }
                } else {
                    // Reject the promise with an error
                    promise.reject("500", "Error configuring react-native-location", e);
                }
            }
        });
```
It happens when the *activity* is null

kkureli, 2021-05-07:
> I also have this crash consistenly
> react-native-location: 2.5.0
>
> the androidProvider in type.td.ts is "auto" | "fused" | "built-in"
> https://www.npmjs.com/package/react-native-location/v/2.5.0
>
> but [...]

hi, could you solve it ?

kkureli, 2021-05-13:
> @kkureli add "activity != null" here:
> 
> if (e instanceof ResolvableApiException && activity != null)
> 
> At least it doesn't crash.

Thanks! 🙏

huy-lv, 2023-04-18:
Thank @pedrolopes10 its work, this repo seem dead