timfpark/react-native-location #84: type 503 warning

sainbayarus, 2019-08-24:
what does meaning "type 503 there is no valid location provider available". Some of devices with android os 9 cannot get location.   RequestPermission returns 'granted' but  when device's location  turned off, this message returns and can not get location.

matt-oakes, 2019-09-19:
If the user has location services turned off, then there's no way for you to get their location. The library tells you this so you can handle it. If you really need to location, I recommend you tell the user why and tell them how to enable location services again.

closed 2019-09-19 by matt-oakes

aymensmurf, 2019-09-24:
How can I get hold of that message {type: 503, message: 'there is no valid location provider available'} so that I can test it?

shiftrtech, 2020-01-24:
There is a way to get this error on my app code, i did try using a `try - catch` but it doesnt works.

Hard-Haad, 2020-06-29:
Is there any way to handle this error, I havnt found anything in the Read me

paddy57, 2021-04-24:
Any update on  how to catch the error ?

mehdinourollah, 2023-02-28:
Why did you close this issue when it's still not resolved ?! 

I am getting this warning despite of having try and catch ...Any help ?