timfpark/react-native-location #86: Error getting user location TypeError: Cannot read property 'startUpdatingLocation' of undefined

ollydixon, 2019-09-01:
Running verison 2.2.1 on Android 9.0 React native version 60.

I get the following error 
`Error getting user location TypeError: Cannot read property 'startUpdatingLocation' of undefined`

When trying to do the following:
`await RNLocation.getLatestLocation()`

Searching around `startUpdatingLocation` looks like an iOS function, why is this being called on Android?

matt-oakes, 2019-09-19:
:wave: Are you manually linking the library or are you using auto-linking? Have you rebuilt the app on Android after installing the library?

ollydixon, 2019-09-19:
This error only occurs with the emulator and only sometimes. Usually fixed after cold boot. Very strange one.