timfpark/react-native-location #87: subscribeToLocationUpdate not working in android 9

J-Ghodasara, 2019-09-02:
I am using your npm package for getting current location but recently i noticed that on android with version 9 devices when GPS is by default off and when i try to turn on gps using requestPermission and then call subscribeToLocationUpdate it is not returning locations. Don't know what the reason is but i have tried on 4 to 5 android 9 devices and its not returning locations. While in other devices with version less than 9 are working fine. I have implemented it as below.
```
const locationSubscription = RNLocation.subscribeToLocationUpdates(
            locations => {
               console.log(locations)
                if (locationSubscription) {
                            locationSubscription();
                        }
                        this.setState({
                            location: locations
                        })
                        this.setCurrentLocation(locations)
            }
        ); 
```

closed 2019-09-03 by J-Ghodasara

Cheg-ya, 2019-09-03:
Did you solve it?
I have exact same issue
Please help!

thecookieorg, 2019-09-07:
I'm having the same issue on iOS (RN: 0.60.5)

karthikpala, 2020-02-25:
Seeing similar issue - did you get this resolved already?