timfpark/react-native-location #89: ./gradlew assembleRelease，error：RNPlayServicesLocationProvider.java:

kevin871223, 2019-09-07:
react-native：0.60.5

 ./gradlew assembleRelease

error：

app\node_modules\react-native-location\android\src\main\java\com\github\reactnativecommunity\location\RNPlayServicesLocationProvider.java:9: 错误: 程序包android.support.annotation不存在
import android.support.annotation.NonNull;
                                 ^
\app\node_modules\react-native-location\android\src\main\java\com\github\reactnativecommunity\location\RNPlayServicesLocationProvider.java:10: 错误: 找不到符号
import android.support.v4.app.ActivityCompat;
                             ^
  符号:   类 ActivityCompat
  位置: 程序包 android.support.v4.app
\app\node_modules\react-native-location\android\src\main\java\com\github\reactnativecommunity\location\Utils.java:5: 错误: 程序包android.support.annotation不存在
import android.support.annotation.Nullable;
                                 ^
\app\node_modules\react-native-location\android\src\main\java\com\github\reactnativecommunity\location\Utils.java:21: 错误: 找不到符号
    public static void emitEvent(ReactApplicationContext context, String eventName, @Nullable Object params) {
                                                                                     ^
  符号:   类 Nullable
  位置: 类 Utils
\app\node_modules\react-native-location\android\src\main\java\com\github\reactnativecommunity\location\RNPlayServicesLocationProvider.java:163: 错误: 找不到符号
            public void onFailure(@NonNull Exception e) {
                                   ^
  符号: 类 NonNull
\app\node_modules\react-native-location\android\src\main\java\com\github\reactnativecommunity\location\RNPlayServicesLocationProvider.java:227: 错误: 找不到符号
            int finePermission = ActivityCompat.checkSelfPermission(context, Manifest.permission.ACCESS_FINE_LOCATION);
                                 ^
  符号:   变量 ActivityCompat
  位置: 类 RNPlayServicesLocationProvider
\app\node_modules\react-native-location\android\src\main\java\com\github\reactnativecommunity\location\RNPlayServicesLocationProvider.java:228: 错误: 找不到符号
            int coarsePermission = ActivityCompat.checkSelfPermission(context, Manifest.permission.ACCESS_COARSE_LOCATION);
                                   ^
  符号:   变量 ActivityCompat
  位置: 类 RNPlayServicesLocationProvider
7 个错误

sorry,my english is poor，can you help me，thanks！

matt-oakes, 2019-09-19:
:wave Can you provide me with the contents of your `android/build.gradle` and `android/app/build.gradle` files? Also, are you using a tool like Jetifier?

handjunyoung, 2019-09-29:
I have experienced the same problem but i solved it.
my solution is using Jetifier.
```
yarn add --dev jetifier
npx jetify
cd ./android && ./gradlew assembleRelease
```

geza909, 2022-08-12:
> I have experienced the same problem but i solved it. my solution is using Jetifier.
> 
> ```
> yarn add --dev jetifier
> npx jetify
> cd ./android && ./gradlew assembleRelease
> ```

it workks!!