timfpark/react-native-location #92: background location in Android 9-10

amirmatin, 2019-09-18:
Hi, 
The background location is working smoothly in Android 8, however, it breaks down after Android 9. 
I have tried using "subscribeToLocationUpdates" and "getLatestLocation" but had no luck...
Any suggestions / solutions ?

Thanks

matt-oakes, 2019-09-19:
:wave @amirmatin 

Can you provide some examples of what you expect to happen and what's happening instead?

amirmatin, 2019-09-19:
Hi Matt,

Sure, so in Android 8 when I use the"subscribeToLocationUpdates" method, while the app goes to the background, it is still able to recieve the correct geolocation info as the location changes so I can use them for turn-by-turn navgation so this is the expected behaviour..
However, on Android 9-10, once the app is minimed, there are no callbacks made upon the user location change. I have tried it on Samsung phone and the Pixel phone running Android 9 and 10 respectivly.

So I tried a different approach where using a [background worker](https://github.com/ocetnik/react-native-background-timer) library in RN , I enforce the new location fetch via the "getLatestLocation" comamnd every 3 seconds however it also returns null values. 
Hope this can shine some light over the issue..
 
Many thanks

matt-oakes, 2019-09-19:
Thanks for the information! This is due to new security policies in Android. This library is also not really designed for background location as, in my opinion, most apps don't really need it.

If you do need it, then I recommend you use this library which supports it really well:

https://github.com/transistorsoft/react-native-background-geolocation

closed 2019-09-19 by matt-oakes