timfpark/react-native-location #96: TypeError: undefined is not an object (near RNLocation.configure({..}).then(...))

naderabdalghani, 2019-10-09:
I have tried rebuilding but the error still persists on iOS. The package works flawlessly on Android though.

Error screenshot:
[image]

Code:
```
  handleGetDirections = () => {
    var directionsData;
    RNLocation.configure({
      distanceFilter: 5.0,
      allowsBackgroundLocationUpdates: true
    })
      .then(() =>
        RNLocation.requestPermission({
          ios: 'whenInUse',
          android: {
            detail: 'fine',
            rationale: {
              title: 'Location permission',
              message: 'Your location is essential to request help',
              buttonPositive: 'OK',
              buttonNegative: 'Cancel'
            }
          }
        })
      )...
```

mileselectric, 2019-10-25:
I'm also getting this same issue.
Works fine on Android but not iOS

shafikgomez, 2019-11-01:
I am getting the same error. Android Works perfect.

gz2jbt, 2019-11-12:
I am experiencing the same error in iOS.  Followed directions and used code from the repo.  Please advise.

naderabdalghani, 2019-11-13:
[react-native-geolocation-service](https://github.com/Agontuk/react-native-geolocation-service) is an alternative solution, which gave me no errors at all

zachlendon, 2019-11-14:
We saw this also, I believe configure in https://github.com/timfpark/react-native-location/blob/d215c8c549330c677af9e30d6752768e6275e169/ios/RNLocation.m needs updated to return a promise.

matt-oakes, 2019-11-17:
`configure` does not return a `Promise`. Just call it once before you use any other method in the library.

closed 2019-11-17 by matt-oakes

zachlendon, 2019-11-17:
@matt-oakes Awesome! I’d propose updating https://github.com/timfpark/react-native-location/blob/9a1c5bf3f683c78de8e82b2820d284e543ef9ab0/example/App.js to reflect   this then.  The reader appears accurate.

matt-oakes, 2019-11-17:
Done!