timfpark/react-native-location #112: How to unsubscribe ToLocationUpdates from another component??

sushil173, 2020-02-05:
// Subscribe
const unsubscribe = RNLocation.subscribeToLocationUpdates(locations => {
  ...
})

// Unsubscribe
unsubscribe();

sultan-arshi, 2020-07-15:
declare a variable some thing like that 

`var unSubFromLocationUpdate: () => void`

`unSubFromLocationUpdate = RNLocation.subscribeToLocationUpdates(locations => {
            // do what ever you want
        })`

if you want `unSubscribeToLocationUpdates` then just call the variable you declared above something like that `unSubFromLocationUpdate ()`

maueli, 2021-08-04:
I am trying to Stop the updates using the variable declared ( unSubFromLocationUpdate() ) like you and the documentation said but never stops..... how I can stop updates? Thanks!

skvaghasiya402, 2023-03-17:
**Hello, please try this trick  -**

_startUpdatingLocation = () => {
    this.locationSubscription = RNLocation.subscribeToSignificantLocationUpdates(
                async ([ locations ]) => {
                  console.log(locations)                                                
                }
          )
  };

_stopUpdatingLocation = () => {
      this.locationSubscription && this.locationSubscription();
      console.log('locations stop')
  };