timfpark/react-native-location #114: Android says my app grabbed the location on background but I didn't

perotta, 2020-02-26:
Hi all,

I'm having a weird situation.

I'm using react-native-location on my app and after a couple of minutes after I fresh install the app, Android sends me a notification saying that my app got the device location on the background. 

The weird thing is that I don't have any background tasks that grabs the location and I'm not opening the screen that uses the location neither.

So why Android is saying that I used the location on background? Anyone have the same situation?

stoneman1, 2020-04-16:
Same issue here. Any update on this?

perotta, 2020-04-16:
The only thing I could find is this: 

https://www.davx5.com/faq/wifi-ssid-restriction-location-permission
and 
https://github.com/react-native-community/react-native-netinfo

> The SSID of the network. May not be present, null, or an empty string if it cannot be determined. On iOS, make sure your app meets at least one of the following requirements. On Android, you need to have the ACCESS_FINE_LOCATION permission in your AndroidManifest.xml and accepted by the user.

since I use NetInfo a lot on my app, it seems that fetching NetInfo state on the background triggers that notifications saying the app grabbed the device's location on background.

It is still unclear to me since I stopped fetching the NetInfo state on the background and the notifications keeps coming up.

I'm using react-native-background-fetch on my app to do stuff on background