timfpark/react-native-location #121: TypeError: null is not an object (evaluating 'nativeInterface.configure')

D3vlin, 2020-03-26:
Hi coders

I am using version 2.5.0
I've tried the example setup in a test app and it works, but when I try it in my app it returns this error.
I am testing it on a physical device with Android 7.0.
This is the configuration that I apply.

```javascript
  componentWillMount() {
    RNLocation.configure({
      distanceFilter: 5.0
    });
    
    RNLocation.requestPermission({
      ios: "whenInUse",
      android: {
        detail: "fine",
        rationale: {
          title: "Location permission",
          message: "We use your location to demo the library",
          buttonPositive: "OK",
          buttonNegative: "Cancel"
        }
      }
    }).then(granted => {
      if (granted) {
        this._startUpdatingLocation();
      }
    });
  }
```

[image]
[image]

D3vlin, 2020-03-26:
I have the same problem with 2.4.0 version

subhani-ali, 2020-07-09:
I have same issue

ReactNativ, 2020-11-16:
add this code

async UNSAFE_componentWillMount() {   
try {
      const granted = await PermissionsAndroid.requestMultiple(
        [PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
        PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION],
        {
          title: "Cool Photo App Location Permission",
          message:
            "Cool Photo App needs access to your location ",
          buttonNeutral: "Ask Me Later",
          buttonNegative: "Cancel",
          buttonPositive: "OK"
        }
      );
      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        console.log("You can use the location");
        return true;
      } else {
        console.log("Location permission denied");
        return false;
      }
    } catch (err) {
      console.warn(err);
    }
}

godfrednanaowusu, 2021-05-29:
any solution to this please, I am trying this in expo, is that a reason, I don't seem to know how to go about this.

Shidooo, 2021-06-29:
I am still experiencing this problem, any solution?

yuyunfrancis, 2022-04-28:
I too have the same error