timfpark/react-native-location #124: How can we detect if GPS permissions denied or not

manjeets12, 2020-04-13:
while using RNLocation.Configure we are getting a promise, can this be used to detect if user pressed "ok" or "No Thanks" from permission alert.
something similar to

```
RNLocation.configure(DESIRED_OPTIONS)
.then((SOME_RESULT_CODE)=>{
   //handle codes and call your functions accordingly
})
.catch(error=>{
 //GPS permission denied
})
```

is there a way to know about this in .then block?

1bastien1, 2020-04-21:
i found on RN documentation a function that check permissions

`
import {check, PERMISSIONS, RESULTS} from 'react-native-permissions';
 
getLocalisation() {
    check(PERMISSIONS.ACCESS_FINE_LOCATION)
      .then((result) => {
        switch (result) {
          case RESULTS.UNAVAILABLE:
            console.log(
              'This feature is not available (on this device / in this context)',
            );
            break;
          case RESULTS.DENIED:
            console.log(
              'The permission has not been requested / is denied but requestable',
            );
            break;
          case RESULTS.GRANTED:
            console.log('The permission is granted');
            RNLocation.getLatestLocation((info) => info);
            break;
          case RESULTS.BLOCKED:
            console.log('The permission is denied and not requestable anymore');
            break;
        }
      })
      .catch((error) => {
        console.log('error getLocalisation : ', error);
      });
  }`

manjeets12, 2020-04-26:
@1bastien1 Yes that method can be used to if permission is granted or not, but it never will never let you know if user peressed "Ok" or "No Thanks" in that particular alert, You may have to call that method again to know about user's last intent.