timfpark/react-native-location #130: RNLocation.getLatestLocation({timeout:6000}) return null after timeout on IOS 13.4

beqramo, 2020-04-30:
After updating my simulator to IOS 13 getLatestLocation returns me null, permissions configured correctly( All granted)

DiegoMac95, 2020-07-02:
@beqramo were you able to solve this?

beqramo, 2020-07-03:
@DiegoMac95  sometimes it happens to me but sometimes not, I don't understand, I think it's bug. I went deep down to the codes but I didn't find any suspicious function and I thing it is on native side bug and I don't know to much about native ios

Hard-Haad, 2020-07-24:
@beqramo experiencing this on android, with react-native-location version 2.5.0 and react-native version 0.62.2. It's like you described sometimes it happens and sometimes it does not, however I have seen that if I change my app's location permission to AlwaysOn then it goes away.