timfpark/react-native-location #131: lint error on location.isFromMockProvider()..

rlyle, 2020-05-08:
I'm trying to build a release react-native application with the react-native-location npm, and we are getting the following error:

* What went wrong:
Execution failed for task ':react-native-location:lint'.
> Lint found errors in the project; aborting build.

  Fix the issues identified by lint, or add the following to your build script to proceed with errors:
  ...
  android {
      lintOptions {
          abortOnError false
      }
  }
  ...

  Errors found:

  F:\Work\Apricity\clients\apricity-care\node_modules\react-native-location\android\src\main\java\com\github\reactnativecommunity\location\Utils.java:52: Error: Call requires API level 18 (current min is 16): android.location.Location#isFromMockProvider [NewApi]
          map.putBoolean("fromMockProvider", location.isFromMockProvider());

Here is a snipet from my build.gradle file in the android directory:

buildscript {
    ext {
        buildToolsVersion = "28.0.3"
        minSdkVersion = 23
        compileSdkVersion = 28
        targetSdkVersion = 28
    }

As you can see my minSdkVersion is 23.

BlackBes, 2020-06-06:
I have the same error. If you can check folder node_modules -> react-native-location -> build.gradle, you can see a minSdkVersion as 16. You can try to switch this to 18.

xaiamov, 2021-11-10:
The same issue