timfpark/react-native-location #136: react-native-location warning:403

kevin871223, 2020-06-20:
react-native:0.58.4

react-native-location warning: 
type:403
Attempted to start updating the location without location permissions. Detail: "gps" location provider requires ACCESS_FINE_LOCATION permission.

thanks!!

Feofilov, 2021-04-12:
> react-native:0.58.4
> 
> react-native-location warning:
> type:403
> Attempted to start updating the location without location permissions. Detail: "gps" location provider requires ACCESS_FINE_LOCATION permission.
> 
> thanks!!

Android API >= 23 Requires an additional step to check for, and request the ACCESS_FINE_LOCATION permission using the PermissionsAndroid API. Failure to do so may result in a hard crash.

You can use something like this before trying to get geolocation:
`
try {
    const granted = await PermissionsAndroid.request(
      PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION,
      {
        title: "Cool App Location Permission",
        message:
          "Cool App needs access to your location ",
        buttonNeutral: "Ask Me Later",
        buttonNegative: "Cancel",
        buttonPositive: "OK"
      }
    );
    if (granted === PermissionsAndroid.RESULTS.GRANTED) {
      console.log("Now you can use geolocation!");
    } else {
      console.log("Location permission denied");
    }
  } catch (err) {
    console.warn(err);
  }
`

qasimmehdi, 2021-04-20:
I am also getting the same error above solution did not help.
```js
   RNLocation.configure({
      distanceFilter: 0, // Meters
      desiredAccuracy: {
        ios: 'best',
        android: 'highAccuracy',
      },
      interval: 1000,
      maxWaitTime: 1000,
    });

   RNLocation.requestPermission({
      ios: 'whenInUse',
      android: {
        detail: 'fine',
        rationale: {
          title: 'Location permission',
          message: 'We use your location to demo the library',
          buttonPositive: 'OK',
          buttonNegative: 'Cancel',
        },
      },
    })
```

This is only happening when desiredAccuracy is highAccuracy.

qasimmehdi, 2021-04-20:
I solved the issue in android.
Add these lines in AndroidManifest.xml
```xml
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
```
Then in android/app/build.gradle add these lines in dependencies
```
implementation "com.google.android.gms:play-services-base:16.0.1"
implementation "com.google.android.gms:play-services-location:16.0.0"
```

bimix, 2022-04-14:
> I solved the issue in android. Add these lines in AndroidManifest.xml
> 
> ```
> <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
> <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
> ```
> 
> Then in android/app/build.gradle add these lines in dependencies
> 
> ```
> implementation "com.google.android.gms:play-services-base:16.0.1"
> implementation "com.google.android.gms:play-services-location:16.0.0"
> ```
> 
> Just follow the installation instructions correctly and it will work.

I already have all of them and still shows the warning