timfpark/react-native-location #138: getLatestLocation() always returns null, subscribeToLocationUpdates never fires

seyaobey, 2020-07-05:
I am encountering the behaviour described in the above title for certain models on samsung devices. I can't tell if it is something wrong I'm doing with the library or it is the device. 
I have properly followed the guidelines to install this library in my react-native application. The user is prompted to grant the location permission and he grants it to the app. Here is the initialisation and the subscription ode:
```
import RNLocation from 'react-native-location';
...

    RNLocation.requestPermission({
      ios: 'whenInUse',
      android: {detail: 'coarse'},
    }).then((granted) => {
      if (granted) {
        RNLocation.getLatestLocation().then((loc) => {
          Alert.alert(JSON.stringify(loc)); // <--- this returns null
        });
        RNLocation.subscribeToLocationUpdates((locations) => {
         // This never fires
          const {latitude, longitude} = locations[0];
          Alert.alert(`${JSON.stringify({latitude, longitude})}`);
        });
      }
    });
```

package.json:
```
"react": "16.11.0",
 "react-native": "0.62.2",
"react-native-location": "^2.5.0"
```

build.gradle
```
buildToolsVersion = "28.0.3"
minSdkVersion = 16
compileSdkVersion = 28
targetSdkVersion = 28
googlePlayServicesVersion  = "17.0.0"
```

My code works perfectly well in an old `Samsung J7 model (2015)`:
```
Android 7.0 (Nougat)
API level 24
```

But the location returned is not correctly in a `Samsung J2 core (2018)`:
```
Android 8.1 (Oreo)
Api level 27
```

In both cases, the location permission is properly granted and gps location is enabled in both devices. 
Thank you for your library. I hope you may help me to find out what I may do wrong or where I should start looking at.

seyaobey, 2020-07-05:
Update on my issue: I was able to solve it, by changing the initialisation for android:
from `coarse` to `fine`. Now it is working as expected in the 2 models. I would just want to know what is the difference between these 2 settings and why it works in the second case (Apologies if this is not the right forum for this discussion). Thank you.

igunultas, 2020-07-27:
According to android docs you can see:
[Google Dev Location Docs](https://developers.google.com/maps/documentation/android-sdk/location)

> Android offers two location permissions: ACCESS_COARSE_LOCATION and ACCESS_FINE_LOCATION. The permission you choose determines the accuracy of the location returned by the API. You only need to request one of the Android location permissions, depending on the level of accuracy you need:
> 
> android.permission.ACCESS_COARSE_LOCATION – Allows the API to use WiFi or mobile cell data (or both) to determine the device's location. The API returns the location with an accuracy approximately equivalent to a city block.
>
> android.permission.ACCESS_FINE_LOCATION – Allows the API to determine as precise a location as possible from the available location providers, including the Global Positioning System (GPS) as well as WiFi and mobile cell data.

HappyCodingLover, 2021-05-18:
@seyaobey I am sorry, but I still get this error even I twitched setting to 'fine' in android.
Can you figure it out why?

velevtzvetlin, 2021-07-01:
@seyaobey I think the issue is you are making those two requests at the same time. 
```
RNLocation.getLatestLocation().then((loc) => {
          Alert.alert(JSON.stringify(loc)); // <--- this returns null
        });
        RNLocation.subscribeToLocationUpdates((locations) => { // <-- this never fires
          const {latitude, longitude} = locations[0];
          Alert.alert(`${JSON.stringify({latitude, longitude})}`);
        });
```
This code is dispatching both the `getLatestLocation` and `subscribeToLocationUpdates` at the same time. I was doing something similar. You can use async methods and call those two sequentially and it should resolve this issue.

ekiziltas, 2023-03-18:
Don't use this repo because of this issue. Please refer to: https://www.npmjs.com/package/react-native-get-location It works fine.