timfpark/react-native-location #154: 🔥 Method addObserver must be called on the main thread

wmonecke, 2021-04-23:
Hi! 👏

I have been using this package for a while until I recently stumbled upon this bug (I did not even update this package).
The error like the title says is just `Method addObserver must be called on the main thread` and appears when calling `getLatestLocation` after getting permission and configuring `RNLocation`.

I am very inexperienced with Java so it is a bit hard for me to fix this bug. I will try to dig into the code.
If anyone is experiencing the same issue, please let me know if you discover a way to solve it.

Cheers!

wmonecke, 2021-04-24:
I changed libraries to `react-native-geolocation-service`

vivekvardhanadepu, 2021-12-15:
using the context rather than activity solved the issue