timfpark/react-native-location #164: Terrible accuracy - physial device Android 10, RN 0.6.4

yogesch, 2021-05-23:
I seem to constantly get terrible accuracy. Even if I ask for fine, I get only coarse. 

The physical device running Android 10 . Connected to computer via USB debugging. React Native version is 0.6.4. 
The initial configuration is : 
```
RNLocation.configure({
  distanceFilter: 10, 
  desiredAccuracy: {
    android: 'highAccuracy',
  },
  androidProvider: 'playServices',
  interval: 5000,
  maxWaitTime: 5000,
});
```

I get permissions like this: 
```
 permission = await RNLocation.requestPermission({
        ios: 'always',
        android: {
          detail: 'fine',
          rationale: {
            title: 'We need to access your location',
            message: 'Your location is needed to check for stores in your area',
            buttonPositive: 'OK',
            buttonNegative: 'Cancel',
          },
        },
      });
```

And I get the location like this 
```
 let location;
 location = await RNLocation.getLatestLocation({timeout: 2000});

```

I don't need to subscribe to location updates, just read it once on clicking a button. 

When I check Google Maps, the location is perfect. 

But on this, it is stuck at some 2000 value.

yogesch, 2021-05-23:
Basically, it only gets the coarse location, never the fine one.

choukrus, 2021-11-18:
Hello, please did you find a good solution to have a better accuracy ?

yogesch, 2021-11-18:
We take a very convoluted approach now. The user is shown whatever address
(based on the coordinates) is returned by the maps api. Then the user has
the choice of editing that address. Again the maps api is used to generate
the coordinates based on the updated address.