timfpark/react-native-location #170: Cant stop updates of RNLocation.subscribeToLocationUpdates

maueli, 2021-08-04:
How can I stop updates? If "stops" == "unsubscribe" so... not work. Even If I define RNLocation.subscribeToLocationUpdates in a constant is not called so I dont have any update. But If I started IT never ends!

skvaghasiya402, 2023-03-17:
**Hello, please try this trick  -**

_startUpdatingLocation = () => {
    this.locationSubscription = RNLocation.subscribeToSignificantLocationUpdates(
                async ([ locations ]) => {
                  console.log(locations)                                                
                }
          )
  };

_stopUpdatingLocation = () => {
      this.locationSubscription && this.locationSubscription();
      console.log('locations stop')
  };

vaidehi-iottive, 2024-07-16:
> this.locationSubscription && this.locationSubscription();
> console.log('locations stop')

Not working any other suggestion