timfpark/react-native-location #179: Failed to build android because of java classes

trickyc0d3r, 2021-12-23:
Task :react-native-location:compileReleaseJavaWithJavac FAILED
/Users/trickyc0d3r/work/delivio/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:5: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
/Users/trickyc0d3r/work/delivio/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:9: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/trickyc0d3r/work/delivio/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:10: error: cannot find symbol
import android.support.v4.app.ActivityCompat;
                             ^
  symbol:   class ActivityCompat
  location: package android.support.v4.app
/Users/trickyc0d3r/work/delivio/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:21: error: cannot find symbol
    public static void emitEvent(ReactApplicationContext context, String eventName, @Nullable Object params) {
                                                                                     ^
  symbol:   class Nullable
  location: class Utils
/Users/trickyc0d3r/work/delivio/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:163: error: cannot find symbol
            public void onFailure(@NonNull Exception e) {
                                   ^
  symbol: class NonNull
/Users/trickyc0d3r/work/delivio/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:227: error: cannot find symbol
            int finePermission = ActivityCompat.checkSelfPermission(context, Manifest.permission.ACCESS_FINE_LOCATION);
                                 ^
  symbol:   variable ActivityCompat
  location: class RNPlayServicesLocationProvider
/Users/trickyc0d3r/work/delivio/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:228: error: cannot find symbol
            int coarsePermission = ActivityCompat.checkSelfPermission(context, Manifest.permission.ACCESS_COARSE_LOCATION);

trickyc0d3r, 2021-12-23:
build.gradle -> `dependencies {
    implementation fileTree(dir: "libs", include: ["*.jar"])
    //noinspection GradleDynamicVersion
    implementation "com.facebook.react:react-native:+"  // From node_modules

    implementation "androidx.swiperefreshlayout:swiperefreshlayout:1.0.0"
    implementation 'androidx.constraintlayout:constraintlayout:1.1.2'
    implementation 'androidx.core:core-ktx:1.6.0'
    implementation 'androidx.core:core:1.6.0'
    implementation 'com.android.support:support-annotations:28.0.0'
    implementation "com.android.support:support-core-utils:28.0.0"
    implementation 'androidx.multidex:multidex:2.0.1'

    implementation 'com.android.installreferrer:installreferrer:2.2'
    implementation "com.google.android.gms:play-services-base:17.6.0"
    implementation "com.google.android.gms:play-services-ads:20.4.0"
    implementation "com.google.android.gms:play-services-location:16.0.0"
    implementation 'androidx.appcompat:appcompat:1.4.0'
    implementation 'androidx.appcompat:appcompat-resources:1.4.0'
    implementation 'com.google.firebase:firebase-messaging:18.0.0'

    debugImplementation("com.facebook.flipper:flipper:${FLIPPER_VERSION}") {
        exclude group:'com.facebook.fbjni'
    }

    debugImplementation("com.facebook.flipper:flipper-network-plugin:${FLIPPER_VERSION}") {
        exclude group:'com.facebook.flipper'
        exclude group:'com.squareup.okhttp3', module:'okhttp'
    }

    debugImplementation("com.facebook.flipper:flipper-fresco-plugin:${FLIPPER_VERSION}") {
        exclude group:'com.facebook.flipper'
    }

    if (enableHermes) {
        def hermesPath = "../../node_modules/hermes-engine/android/";
        debugImplementation files(hermesPath + "hermes-debug.aar")
        releaseImplementation files(hermesPath + "hermes-release.aar")
    } else {
        implementation jscFlavor
    }
}`

trickyc0d3r, 2021-12-23:
can some one help fix this?

nikita-usichenko, 2022-01-07:
@trickyc0d3r hi, did you find the answer?

Ahmad0811, 2022-01-16:
Run in your project root 
`
npx jetify
`

tomdionysus, 2022-02-14:
> @rickyc0d3r & @nikita-usichenko Run `npx jetify` in your project root

Can confirm this fixes 'package androidx.annotation.Nullable not found' error when building Android Studio - thanks!

ShadowClone26, 2022-08-18:
> @rickyc0d3r & @nikita-usichenko Run `npx jetify` in your project root

thanks this works for me

NguyenPhuongNam99, 2022-09-18:
@Ahmad0811 hi. I faced problem when I run npx jetify : this is error : jetifier is deprecated. future versions of react-native cli may not run it by default.

Anudnya93, 2022-10-16:
npx jetify  Worked for me.

if you get this:

Please consider installing 'jetifier' package before running 'jetify' command!    ----then run :

yarn add jetifier  and hit npx jetify again. Thanks