timfpark/react-native-location #191: Requesting "always" permission when user has previously selected "whenInUse" causes request to hang (promise never resolves) or deprecate this library

iway1, 2022-04-13:
Since iOS 13 apps are only allowed to request location permission a single time. If a user has previously selected to share location when in use, the ios code will attempt to ask for always permission even though it's impossible to ask for permissions a second time.

SO, the promise never resolves. In my case, my app froze forever because I am performing a location services check and waiting for it to resolve before navigating the user to the main screen of the app.

Pretty clear that this library isn't being actively maintained. If it's not going to be fixed, then please for the love of god update the read me to point out that this library doesn't work properly with iOS 13+ and that it's not being maintained and that it should not be used.

NaitikDartexon, 2022-10-06:
Same in my case this heppens in android after give the allow location permission when i put the apk in background and open again second time it ask for the location, some time while i close the application and open at that time still it ask for permission. after second time permission can see the location pop-up

NaitikDartexon, 2022-10-10:
android: {
        detail: 'fine'
      }
      
      i added this for request permission now it can't ask twice permission pop-up for me