timfpark/react-native-location #198: Failed to Build

AlexMercedCoder, 2022-09-10:
I'm reviewing this tutorial: https://blog.logrocket.com/react-native-geolocation-a-complete-tutorial/

Everything goes fine till I need to install react-native-location

- I can install the library fine
- The link command is no longer part of the CLI but I get errors that it has not been linked

I figured maybe I need to run `npx react-native run-android` to get it to link

but then I get the following output

```
info Starting JS server...
info Installing the app...
<===----------> 25% CONFIGURING [873m> Task :react-native-location:compileDebugJavaWithJavac FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 8.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/7.5.1/userguide/command_line_interface.html#sec:command_line_warnings
33 actionable tasks: 2 executed, 31 up-to-date
warning: [options] source value 7 is obsolete and will be removed in a future release
warning: [options] target value 7 is obsolete and will be removed in a future release
warning: [options] To suppress warnings about obsolete options, use -Xlint:-options.
/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:5: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:9: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:10: error: cannot find symbol
import android.support.v4.app.ActivityCompat;
                             ^
  symbol:   class ActivityCompat
  location: package android.support.v4.app
/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/Utils.java:21: error: cannot find symbol
    public static void emitEvent(ReactApplicationContext context, String eventName, @Nullable Object params) {
                                                                                     ^
  symbol:   class Nullable
  location: class Utils
/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:163: error: cannot find symbol
            public void onFailure(@NonNull Exception e) {
                                   ^
  symbol: class NonNull
/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:227: error: cannot find symbol
            int finePermission = ActivityCompat.checkSelfPermission(context, Manifest.permission.ACCESS_FINE_LOCATION);
                                 ^
  symbol:   variable ActivityCompat
  location: class RNPlayServicesLocationProvider
/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/react-native-location/android/src/main/java/com/github/reactnativecommunity/location/RNPlayServicesLocationProvider.java:228: error: cannot find symbol
            int coarsePermission = ActivityCompat.checkSelfPermission(context, Manifest.permission.ACCESS_COARSE_LOCATION);
                                   ^
  symbol:   variable ActivityCompat
  location: class RNPlayServicesLocationProvider
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
7 errors
3 warnings

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-location:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
> Run with --stacktrace option to get the stack trace.
[... 227 lines cut ...]
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
java.lang.StackOverflowError (no error message)

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 2s

    at makeError (/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/execa/index.js:174:9)
    at /home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/execa/index.js:278:16
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async runOnAllDevices (/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/@react-native-community/cli-platform-android/build/commands/runAndroid/runOnAllDevices.js:109:5)
    at async Command.handleAction (/home/alexmerced/development/developmentwork/logrocket/react-native-review/ReactNativeGeolocation2/node_modules/@react-native-community/cli/build/index.js:142:9)
```

haidar4243, 2022-09-13:
facing the same issue. did you find any solution yet

tkyr-hh, 2022-09-15:
I had the same issue but `npx jetifier` solve my issue, maybe the main difference is that the `jetifier` package was installed in my dependency list.

NguyenPhuongNam99, 2022-09-17:
hello @tkyr-hh  I faced problem when I use to npx jetify ,, The error is : Jetifier is deprecated. Future versions of react-native CLI may not run it by default. Please help me

tkyr-hh, 2022-09-18:
@NguyenPhuongNam99  It is just a deprecation working my friend, you can use the jetifier to convert the package but it is not recommended because it has been deprecated.

NguyenPhuongNam99, 2022-09-18:
@tkyr-hh thank you for your response. How can fix it? . I am not running project. Please help me

NguyenPhuongNam99, 2022-09-18:
@tkyr-hh and other resolve problem ?

tkyr-hh, 2022-09-18:
@NguyenPhuongNam99 show me the logged error please

NguyenPhuongNam99, 2022-09-18:
thanks @tkyr-hh , Log error is: Jetifier is deprecated. Future versions of react-native CLI may not run it by default.

Coder-hrl, 2023-02-10:
l also solve it on run  `run  npx jetifier`

timothyerwin, 2023-02-24:
Same problem here. I'm using RN 0.70.5.

When I try the `npx jetify` I just get

```sh
Your Princess is in Another Castle!

  Please consider installing 'jetifier' package before running 'jetify' command!

Exiting...
```

Besides it's obsolete now...is this package also obsolete? lol

helloworq, 2023-08-22:
> I had the same issue but `npx jetifier` solve my issue, maybe the main difference is that the `jetifier` package was installed in my dependency list.

thanks it works for me

avishek-dev, 2023-08-25:
I am facing this when I have made `compileSdkVersion` and `targetSdkVersion` to **33**
```
buildscript {
    ext {
        buildToolsVersion = "31.0.0"
        minSdkVersion = 21
        compileSdkVersion = 33
        targetSdkVersion = 33
        ndkVersion = "21.4.7075529"
    } 
....
....
}
```

When I'm trying to generate a **release APK** that's when I got this error.

```
 > Task :react-native-location:compileDebugJavaWithJavac FAILED
FAILURE: Build failed with an exception.

*  What went wrong:
Execution failed for task ':react-native-location:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

```

Any suggestion from any on how to solve this issue? Any suggestion or help will be appreciated since I've been blocked for a long time on time.