timfpark/react-native-location #201: RNLocation functions issue

RoshaanAli, 2022-10-15:
hello @timfpark 
You made a great module for working with bglocation. Actually suddenly it stops working.
[image]
**TIA**

rexcode, 2022-10-15:
Hi Tim,

got same issue - working smooth for many months  suddenly start getting  this issue

[image]

kindly check and guide

using
react-native": "0.63.2"
react-native-location": "^2.5.0"

ereenyessamhanna, 2022-10-19:
Check your build.gradle inside android that googlePlayServicesVersion is not '+'
buildscript {
        ext {
            googlePlayServicesVersion = "your google play service version" //  expected to be your google play service version
        }
}

you can see that the last release note is at October 13, 2022, there is a change inside the location which maybe it's the cause of this crash check this link: 
https://developers.google.com/android/guides/releases

bibhuti9, 2023-01-30:
I install this package and it through this error while installing react native app 

[image]

ahmadmalindo, 2023-06-08:
> I install this package and it through this error while installing react native app
> 
> [image]

problem in firebase not in react native location

bibhuti9, 2023-06-08:
issue now resolved,

run npx jetifier before run react native project